dotnet/iot #668: Exception while reading GPIO pin

mkramar, 2019-08-10:
Hi guys!

I'm on Ubuntu 18.04.2 and I'm trying to run the following code:

```
class Program
{
	static void Main(string[] args)
	{
		var gpioDriver = new LibGpiodDriver(2);
		var pinNum = new PinNumberingScheme();
		var gpio = new GpioController(pinNum, gpioDriver);

		gpio.OpenPin(17);
		
		var recv = gpio.Read(17);	// this line crashes
	}
}
```

gpio.Read gives me the following exception:

```
Exception has occurred: CLR/System.IO.IOException
An unhandled exception of type 'System.IO.IOException' occurred in System.Device.Gpio.dll: 'Error while reading value from pin: 17, error: 1'
   at System.Device.Gpio.Drivers.LibGpiodDriver.Read(Int32 pinNumber)
   at System.Device.Gpio.GpioController.Read(Int32 pinNumber)
   at Test1.Program.Main(String[] args) in /home/misha/0-temp/Test1/Program.cs:line 26
```

If I try to read that pin number using **gpioget** it works just fine and returns 1:

```
gpioget 2 17
1
```

Any ideas what might cause the problem?

ZhangGaoxing, 2019-08-10:
try `sudo`. Error 1 means operation not permitted

mkramar, 2019-08-10:
> try `sudo`. Error 1 means operation not permitted

Zhang, `gpioget` seems to do just fine without `sudo`.
By the way, when I switch from 'LibGpiodDriver` to `SysFsDriver` it also works fine

Frankenslag, 2019-08-11:
> try sudo. Error 1 means operation not permitted
> 
> Zhang, gpioget seems to do just fine without sudo.
> By the way, when I switch from 'LibGpiodDrivertoSysFsDriver` it also works fine

I don't think that 1 means not permitted. I believe that the LibGpioD call will just return zero for success and 1 for any other problem. 

What hardware are you using ?

joperezr, 2019-08-12:
cc: @buyaa-n as FYI as she is who wrote libgpiod driver. Are you aware what error 1 might mean? Is the driver supposed to work without sudo?

@mkramar if you try to run the same program with sudo, does the program run fine?

buyaa-n, 2019-08-12:
it should work without sudo, @mkramar are you sure your intended GPIO chip id is 2?

joperezr, 2019-08-12:
>are you sure your intended GPIO chip id is 2?

It looks like so, check his initial comment he tried running the gpiod commands, use number 2 as well, and that worked so I don't think that is the issue.

buyaa-n, 2019-08-12:
oh i see @joperezr

@mkramar did you tried with setting pinMode explicitly, by default it should be Input but not sure if we ever tested in Ubuntu, just in case try: `gpio.OpenPin(17, PinMode.Input);`

Question: what device you are using?

krwq, 2019-08-12:
Are you sure libgpiod is avaialble by default on `Ubuntu 18.04.2`?

@mkramar could you share per @buyaa-n's question what board it is?

Also please share following:

- `dotnet --info`
- version of `System.Device.Gpio` you are using (it should be visible in csproj file)
- `cat /proc/cpuinfo`
- check if `gpioinfo`/`gpiodetect` command returns any results (might require sudo)

mkramar, 2019-08-12:
Thanks everyone for the comments!

The board is https://up-board.org/upcoreplus/specifications/

I'm away from my machine so I cannot answer other questions right now.
Yes, `libgpiod` is installed.
Yes, I can run `gpiodetect`, `gpioget` etc. without `sudo` and it works fine.

Idea of setting pin mode sounds good to me. Will try later.

Ellerbach, 2020-12-22:
@mkramar do you still have an issue?

krwq, 2020-12-22:
I'll close this as this most likely is solved. @mkramar feel free to re-open if you see any more issues.

closed 2020-12-22 by krwq