dotnet/iot #674: RegisterCallbackForPinValueChangedEvent giving multiple events

LeivoSepp, 2019-08-15:
What I have:
4 inputs with the buttons and 2 inputs with a relay to switch them on and off.

Two big issues with the event handler are following:
1. Current driver doesnt have an option to set Debounce Timeout for a pin. This functionality was exist in Windows IoT core world where I just set the option pin.DebounceTimeout = TimeSpan.FromMilliseconds(50);
How to solve the situation where I am getting few events in very short period of time but I have to count them like one event? This is normal situation with any button.
2. Current driver has different event handlers for Rising and Falling. In Windows IoT core world I was using the code button.ValueChanged += btn_ValueChanged;
If I register pin event handler for rising and falling then I will get randomly events called out. No pattern can be figured out. It could be related also to previous issue related to bounce timeout.

krwq, 2019-08-15:
@LeivoSepp the easiest solution for this would be hardware: simply add tiny capacitor between ground and your pin - that should remove all high frequency signals (or oscillation during short period of time when you press the button). The larger the capacitor the longer you will have to press the button for.

From software solution I don't have exactly what you're asking for but I have something I wrote some time a ago and it might help in your case and can be adjusted to do exactly what you wanted (I've had similar problem with ADC where it was oscillating around the threshold I was waiting for):

```csharp
        static void WaitForSignal(Func<bool> condition, int repeatsToAccept = 5, int delayMicroseconds = 50, bool allowThreadYieldDuringDelay = true)
        {
            int samplesLeft = repeatsToAccept;
            while (samplesLeft > 0)
            {
                if (condition())
                {
                    samplesLeft--;
                }
                else
                {
                    samplesLeft = repeatsToAccept;
                }

                DelayHelper.DelayMicroseconds(delayMicroseconds, allowThreadYield: allowThreadYieldDuringDelay);
            }
        }
```

DelayHelper you can copy from here https://github.com/dotnet/iot/blob/master/src/devices/Common/System/Device/DelayHelper.cs

The way this works it waits for 5 consecutive signals (measured with 50us delay). If you get 4 signals and one "`false`" signal it will start counting again until you get exactly 5 positive signals.

Here is how I use it:

```csharp
WaitForSignal(() => adc.Read(0) > 900);
// in your case it might be something like:
WaitForSignal(() => (bool)gpio.Read(5)); // or negate to wait for Low
```

you might need to adjust the code as needed or modify it for timing.

I'm sure we will eventually have some better solution for this (either something similar to above or something closer to what you proposed here).

Unfortunatelly this solution will read in a loop instead of using hardware interrupts or some solution provided by OS (although this might actually have better response for closer to real-time usage).

joperezr, 2019-08-16:
@LeivoSepp I agree having debounce timeout would be helpful. Current implementation would force developer to do their own debounce logic, so this issue is not a shipblocker for 3.0 release, but it would be nice to add for the future. @LeivoSepp are you interested in submitting a formal api proposal for this?

LeivoSepp, 2019-08-18:
Thanks for the DelayHelper and code sample. 
I did a test on my table and it works. After switching to my real environment it getting again false-events.
After lot of testing I figured out that I have some power cables (ventilation motor) very close to Raspberry cables. When the ventilation motor will be turned on then power cable interference my Raspberry and the pin's will get fire an event. Turning off the motor, doesn't fire events.
One of my pin is low by default and this pin will fire always falling event. 
Second pin is high and it will fire always rising event.
Both pins mode is InputPullUp.
**In conclusion:** 
If the pin is low it will fire only falling event caused by power cable interference. Logic says, that before falling it has to be high but it isn't. 
And also opposite is true. If the pin is high, it will fire only rising event caused by power cable interference. Again logic says that before rising it has to be low, but it isn't. 

**Exactly same physical setup** - power cables and Raspberry cables are close to each other - worked like a charm with Windows IoT core.

P.S. Adding capacitor doesn't work correctly because then the voltage is falling or rising "linearly and slowly". There is a level, where GPIO has to change its level and if you are very close on that level then sometimes it fires still two or more events.
I tried different capacitors but this didn't solve the issue caused by the power cables.

krwq, 2019-08-19:
@LeivoSepp thanks for testing! I've also noticed some random events on my box but only up to first 2 seconds after opening the port, after that there are no issues - I've read various threads on the Internet and people have reported this issue on Linux. If I have some spare time I'll try to see if there is some solution for that (or perhaps you can send PR if you got some time to read about it)

Generally raising/falling detection with events won't be 100% reliable because the events get raised independently of the state of the pin value (meaning that when we detect raising and falling event your code might be executed after you get both events or in other words time of raising the event is not the same as when event handler gets executed) - this is matching what we do on Windows and other events. Generally I wouldn't rely on events if changes to the pin happen faster than ~10 ms or when your cpu is busy (minimum Thread.Sleep timeout) - for that only tight loop will be the most reliable.

krwq, 2023-02-02:
[Triage] Fixed by https://github.com/dotnet/iot/pull/1720

closed 2023-02-02 by krwq