dotnet/iot #681: Can you please implement the driver for MCP3204?

pankaj-nikam, 2019-08-22:
Hi team,

Thank you for contributing towards the IoT devices. I would like to have a driver for MCP3204. It would be great if it is implemented. At least a base class from which these devices can inherit and use it. For example a base class of MCP32XX can help us implement the classes for MCP3204, MCP3208 etc. somewhat like what GPIO Zero library.

In case you want the datasheet for MCP3204, it can be found in the link below:

http://ww1.microchip.com/downloads/en/DeviceDoc/21298e.pdf

Thanks in advance.

Frankenslag, 2019-08-23:
@joperezr 
Happy to look at this if you want to assign me. 

@pankaj-nikam 
Not sure which GPIO Zero library you mean. I believe that we would want to keep bindings for these ADC's looking like bindings for ADC's that we already  have.

krwq, 2019-08-23:
@Frankenslag yes please try to make it as similar to Mcp3008 as possible (possible even some of the code can be reused)

pankaj-nikam, 2019-08-24:
Hi @Frankenslag, 

Sorry for not being clear, what I meant to say is the way GPIOZero is implemented, the base class is for MCP3XXX where the derived classes have the correct implementation - this can help code be clean. Thats all what I wanted to convey. In case you are wondering the docs of the same, please check: https://gpiozero.readthedocs.io/en/stable/api_spi.html#base-classes

Thanks.

Frankenslag, 2019-08-24:
I think that it would be possible to have the ADC's derived as you describe but I think that first I will need to get the Mcp3204 and 3208 working and then discuss if we take the effort to include the others. We would have to be careful as people may be already using the 3008.

pankaj-nikam, 2019-08-26:
Hi @Frankenslag , I agree with you. We should not break the existing code. What we can do is create a super base class which handles the common things, then a 32xx and a 30xx which inherit from it; so that the 10 bit and 12 bit things can be handled differently. As the new devices get added, it will still be working. What do you think?
Thanks.

krwq, 2019-09-13:
@pankaj-nikam @Frankenslag I'm closing this since PR is merged, feel free to reopen if you think there is anything left out or any follow up work

closed 2019-09-13 by krwq