dotnet/iot #686: Investigate use of .NET Core 3 MemoryMappedFiles

Frankenslag, 2019-08-24:
Can I suggest that Investitgate he use of the .net core memorymappedfiles instead of using interrop in the raspberry pi driver and any other that may benefit.

[image]

Frankenslag, 2019-08-26:
You can assign it to me If you want. It was originally suggested by @krwq I think. I am guessing it is for after 3.0 as it would be a bit risky because all our current testing has been done with interrop.

Frankenslag, 2019-08-27:
@joperezr @krwq

It looks like MemoryMappedFile is not going to work with /dev/mem and /dev/gpiomem until 5.0 which is a shame.

 This appears to be an issue with using FileStream internally and the checks that FileStream makes with device files causes the exception `Stream does not support seeking`

[Please see here](https://github.com/dotnet/corefx/issues/30686)

Frankenslag, 2019-08-28:
> I see, thanks for looking into it @Frankenslag, is there an issue for that in corefx already?

Yes. It’s # 30686 in corefx

krwq, 2025-02-06:
[Triage] @Frankenslag closing because this doesn't seemed to have gained enough attention. Please re-open or just send a PR if you feel like improving this.

closed 2025-02-06 by krwq