dotnet/iot #687: PlatformNotSupportedException is raised when access PinValue

maxim-tkachenko, 2019-08-24:
I'm getting PlatformNotSupportedException when access PinValue:
`var value = PinValue.High;`

`System.PlatformNotSupportedException: Operation is not supported on this platform.
   at System.Device.Gpio.PinValue.get_High()`
Why does it happen? `PinValue` doesn't have anything platform specific. I'm running the code on Mac.

maxim-tkachenko, 2019-08-26:
hi @joperezr, thanks for the answer, didn't know about the tool for mac. Yes, I have a dev environment on mac but run on the raspberry. I've mocked GpioController for debug builds, but since another base library types (like PinValue) doesn't implement any specific interfaces, there is no way to do it. The only way currently I see is to create some adapter class which will work with all library specific types. Is it possible do not throw the not supported exception for the common types of all platforms? Maybe I'm not the only one who develop on the mac :)

maxim-tkachenko, 2019-08-27:
unfortunately, operator overloads throw the same PlatformNotSupportedException

maxim-tkachenko, 2019-08-28:
yeah, that's what I meant when was writing about adapter class. And it's only way currently I see to achieve my needs.

DavidKarlas, 2019-12-23:
I ran into same problem today, with `GpioDriver` constructor...
If you don't care about what happens on macOs, why did you make tool that injects exceptions?

My use case is... I want to write custom GpioDriver that talks with Arduino(with custom .ino) over serial and expose Arduino pins, so I can do whole logic in C# on Mac/Linux/Windows...

DavidKarlas, 2019-12-23:
For anyone who also runs into this issue, do `cp ~/.nuget/packages/system.device.gpio/1.0.0/runtimes/linux/lib/netstandard2.0/* ~/.nuget/packages/system.device.gpio/1.0.0/lib/netstandard2.0/` as workaround and rebuild your project.

pgrawehr, 2019-12-24:
@joperezr Maybe we could include a simulation implementation instead of the NotSupported implementation? In a first version, this would probably just act as a Null device, but could be extended up to something that reads diagrams and simulates them.

madameczek, 2020-10-08:
I faced PlatformNotSupportedException as well. Target linux-arm platform Raspberry Pi 4 (Raspbian buster), .NET Core on Pi 3.1.301.
Firstly on OpenPin(). 
Then replaced System.Device.Gpio 1.0.0 to 1.1.0-prerelease.20276.1.
Now the exception is thrown on creating GpioController. It is detected by debugger, but doesn't fall into catch statement.
A pin can be opened and written, at least.

Ellerbach, 2020-10-09:
@madameczek this is related to latest RPI4 raspbian update. This has been fixed few days ago. 
Please use the lastest should be 20508.1 and that should solve your issue.

madameczek, 2020-10-09:
Thx for info. After OS upgrade, issue is solved.

Ellerbach, 2020-10-09:
Great, thanks @madameczek, so closing the issue!

DavidKarlas, 2020-10-09:
@Ellerbach Original bug is about exception on Mac which is not fixed as far as I understand

Ellerbach, 2020-10-09:
@DavidKarlas so reopening it!

pgrawehr, 2020-10-09:
@Ellerbach : I have no mac, so I cannot verify,  but I do believe the issue is solved with the removal of the platform dependent assemblies. The main problem was that for any unsupported OS, an assembly was used that consisted only of PNSE. 
Applications such as the one mentioned above,  with an Arduino on Mac or Windows are possible now (PR pending)

joperezr, 2020-10-09:
What @pgrawehr says above is correct. The PNSE issue has been fixed post 1.0 when we removed platform specific implementations. Closing again 😄

closed 2020-10-09 by joperezr