dotnet/iot #688: Device emulator for test

EuroEager2008, 2019-08-25:
I am writing some code using System.Device.Gpio and it would be nice to be able to test with some "mock emulator", preferably acting similar to the real device.
Possible?
Existing?

krwq, 2019-08-25:
We currently don't have any simulators but many devices should be easy to change to mock the devices.

Is there any specific device you'd like to mock? If GpioController you can implement this abstract class: https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/GpioDriver.cs and then pass it to GpioController constructor

krwq, 2023-02-02:
[Triage] Closing as duplicate of https://github.com/dotnet/iot/issues/1828

closed 2023-02-02 by krwq