dotnet/iot #698: led-blink sample does not run

JPThorne, 2019-09-03:
**Description**

The dockerfile for the led-blink sample does not appear to be in working order.

**Steps to reproduce**

Using the commands listed on the README for the led-blink sample:
```
cd samples
cd led-blink
docker build -t led-blink .
docker run --rm -it -v /sys:/sys led-blink
```
The docker build command instead errors with:

`COPY failed: no source files were specified`

Having commented out the step `COPY *.config .` The following error occurs: 

```
Microsoft (R) Build Engine version 16.2.32702+c4012a063 for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Skipping project "/app/System.Device.Gpio.csproj" because it was not found.
  Skipping project "/app/System.Device.Gpio.csproj" because it was not found.
  Restore completed in 34.65 ms for /app/led-blink.csproj.
/usr/share/dotnet/sdk/2.1.801/Microsoft.Common.CurrentVersion.targets(1875,5): warning : The referenced project 'System.Device.Gpio.csproj' does not exist. [/app/led-blink.csproj]
Program.cs(6,14): error CS0234: The type or namespace name 'Device' does not exist in the namespace 'System' (are you missing an assembly reference?) [/app/led-blink.csproj]
The command '/bin/sh -c dotnet publish -c release -o out' returned a non-zero code: 1
```

**Expected behavior**
I expected the dockerfile to at least build, and a runnable image to be created. 

**Actual behavior**

As described above the dockerfile is erroring.

**Versions used**

Output of `dotnet --info`: 

```
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100-preview6-012264
 Commit:    be3f0c1a03

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.17134
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.0.100-preview6-012264\

Host (useful for support):
  Version: 3.0.0-preview6-27804-01
  Commit:  fdf81c6faf

.NET Core SDKs installed:
  2.1.202 [C:\Program Files\dotnet\sdk]
  2.1.302 [C:\Program Files\dotnet\sdk]
  2.1.401 [C:\Program Files\dotnet\sdk]
  2.1.402 [C:\Program Files\dotnet\sdk]
  2.1.504 [C:\Program Files\dotnet\sdk]
  2.1.507 [C:\Program Files\dotnet\sdk]
  2.1.508 [C:\Program Files\dotnet\sdk]
  2.2.104 [C:\Program Files\dotnet\sdk]
  3.0.100-preview6-012264 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0-preview6.19307.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.0.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.3-servicing-26724-03 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0-preview6-27804-01 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.0-preview6-27804-01 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
```

Quite new to this library so I may of course be missing something!
Thanks.

joperezr, 2019-09-03:
Hello @JPThorne thanks a lot for filing this. We did recently break this because we moved all projects (includding the samples) to use P2P references but we didn't take into consideration that samples should probably still reference nuget in order to work well in the docker scenarios. We can fix this by reverting back to use package reference for the samples.

krwq, 2019-09-03:
@joperezr can we add some props in msbuild in the root to make the samples use P2P when build inside the repo and nuget package when copied over elsewhere?

joperezr, 2019-09-03:
We probably can but the main point of the samples was for them to be able to copy paste them outside the repo and have them work just as regular projects would, so in this case I think it's fine to bring pack package reference just for these guys.

krwq, 2019-09-03:
@joperezr so we should at least not build them by default on the CI - otherwise we will have delay broken build whenever we adjust APIs in System.Device.Gpio

joperezr, 2019-09-03:
This was a problem in the past because API shape was changing a lot, but I don't expect this to be a problem in the future given that we are stabilizing now.

krwq, 2019-09-03:
@joperezr that's true for current APIs but I expect we will add some new stuff in the next version

JPThorne, 2019-09-05:
Thanks for the feedback, I'll try to get it working locally so long then.

krwq, 2023-02-02:
[Triage] Seems like a dup of https://github.com/dotnet/iot/issues/764

closed 2023-02-02 by krwq