dotnet/iot #701: Compilation issue with SixLabors library.

Frankenslag, 2019-09-05:
When building within visual studio I receive the following.

```
Error	NU1605	Detected package downgrade: Microsoft.Net.Compilers.Toolset from 3.1.0 to 3.1.0-beta1-19127-06. Reference the package directly from the project to select a different version. 
 Ssd13xx.Samples -> SixLabors.ImageSharp.Drawing 1.0.0-beta0007 -> Microsoft.Net.Compilers.Toolset (>= 3.1.0) 
 Ssd13xx.Samples -> Microsoft.Net.Compilers.Toolset (>= 3.1.0-beta1-19127-06)	
```
It looks like SixLabors have added a dependency to Microsoft.Net.Compilers.Toolset last night. 

Should I create a PR to raise the compiler version to 3.1.0 or is there something more subtle that I should do ?

krwq, 2019-09-05:
cc: @joperezr

joperezr, 2019-09-05:
This has been fixed in latest master, I fixed this yesterday. Try syncing the code and building again.

joperezr, 2019-09-05:
Closing as this has been fixed. @Frankenslag feel free to reopen if you still hit this after updating to latest in master.

closed 2019-09-05 by joperezr