dotnet/iot #702: GPIO output not working on relay in raspbian

Codewizard, 2019-09-07:
**Describe the bug**

raspbian PI3 trying to create interfaces, powering a relay, all commands execute fine but it seems that the relay will not power, I tested it in python and the relay itself works fine.

**Steps to reproduce**

```csharp
// put sample code here
```
 public class GpioAccess
    {

        private PinNumberingScheme boardLayout { get; set; } = PinNumberingScheme.Logical;
        public GpioAccess() {}
        public GpioAccess(PinNumberingScheme pinLayout) { this.boardLayout = pinLayout; }
        public void SetBoardLayout(PinNumberingScheme pinLayout) { this.boardLayout = pinLayout; }
        public enum ResistorDirection {
            Up,
            Down
        }

        public void Write(int pinId,PinValue value)
       {
            using (GpioController controller = new GpioController(boardLayout, new RaspberryPi3Driver()))
            {
                if (controller.IsPinOpen(pinId))
                {
                    controller.ClosePin(pinId);
                }
                controller.OpenPin(pinId, PinMode.Output);
                controller.Write(pinId, value);

            }
        }
        public PinValue Read(int pinId) {
            using (GpioController controller = new GpioController(boardLayout, new RaspberryPi3Driver()))
            {
                if (controller.IsPinOpen(pinId))
                {
                    controller.ClosePin(pinId);
  
                }
                controller.OpenPin(pinId, PinMode.Output);
                return controller.Read(pinId);
            }
        }
        public PinValue Read(int pinId,ResistorDirection direction)
        {
            using (GpioController controller = new GpioController(boardLayout, new RaspberryPi3Driver()))
            {
                if (controller.IsPinOpen(pinId))
                {
                    controller.ClosePin(pinId);
                  }
                controller.OpenPin(pinId, direction == ResistorDirection.Up ? PinMode.InputPullUp : PinMode.InputPullDown);
                return controller.Read(pinId);

            }
        }

    }

**Expected behavior**

Running similar code with python, the relay will activate when supplied a 1 and deactivate when supplied a 0

**Actual behavior**

Functions do not return error, but relay does not activate.

**Versions used**
1.0.0-prerelease.19413.1

Add following information:

- `dotnet --info` on the machine being used to build
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- Version of `System.Device.Gpio` package1 .0.0-prerelease.19413.1
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

build machine
.NET Core SDK (reflecting any global.json):
 Version:   2.2.101
 Commit:    236713b0b7

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18362
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\2.2.101\

Host (useful for support):
  Version: 2.2.0
  Commit:  1249f08fed

.NET Core SDKs installed:
  1.0.0 [C:\Program Files\dotnet\sdk]
  1.0.3 [C:\Program Files\dotnet\sdk]
  2.1.202 [C:\Program Files\dotnet\sdk]
  2.1.400 [C:\Program Files\dotnet\sdk]
  2.1.401 [C:\Program Files\dotnet\sdk]
  2.1.402 [C:\Program Files\dotnet\sdk]
  2.1.403 [C:\Program Files\dotnet\sdk]
  2.1.500 [C:\Program Files\dotnet\sdk]
  2.1.502 [C:\Program Files\dotnet\sdk]
  2.1.503 [C:\Program Files\dotnet\sdk]
  2.1.505 [C:\Program Files\dotnet\sdk]
  2.1.508 [C:\Program Files\dotnet\sdk]
  2.2.101 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.5 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.5 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 1.0.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 1.1.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.0.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.3-servicing-26724-03 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download

Run machine
.NET Core SDK (reflecting any global.json):
 Version:   2.2.102
 Commit:    96ff75a873

Runtime Environment:
 OS Name:     raspbian
 OS Version:  9
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet/sdk/2.2.102/

Host (useful for support):
  Version: 2.2.1
  Commit:  878dd11e62

.NET Core SDKs installed:
  2.2.102 [/home/pi/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.2.1 [/home/pi/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.2.1 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.2.1 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download

krwq, 2019-09-07:
@Codewizard try creating 1 GpioController instance per class and see if that makes any difference. GpioController closes pin when it's disposed (which is implied by using `using` statement)

Codewizard, 2019-09-07:
I tried that by making the class a singleton and dropping the controller in a a static instance variable, same behavior, it does not turn the relay on, it kicks the light for that GPIO off, and never turns it back on.

krwq, 2019-09-10:
@Codewizard can you share your new code? I've tried realys without any issues on Raspbian. One thing to check is if the code doesn't require sudo.

Ellerbach, 2020-12-22:
@Codewizard do you still have an issue?

krwq, 2020-12-28:
@Codewizard feel free to reopen if you still seeing any issues. I'm closing this since we've tried relays on several occasions without any issues

closed 2020-12-28 by krwq