dotnet/iot #706: Which is the architecture of .NETCore?

matsujirushi, 2019-09-10:
In [requirements](https://github.com/dotnet/iot/blob/master/samples/README.md#requirements).

1. .NET Core 2.1. .NET Core 3.0 is required for ARM64 devices.
2. .NET Core is supported on Raspberry Pi 2B as well as 3 and 4 models (ARMv7/v8).

Which is the architecture of .NETCore for Raspbian on Raspberry Pi? ARM32 or ARM64?

Frankenslag, 2019-09-10:
For standard Raspbian it's Arm32

krwq, 2019-09-10:
.NET Core depends on OS you're using. If you install Ubuntu then ARM64 is also an option but Raspbian (binary) is I believe only ARM32. Source dist might have larger support though and I think there might be a way to make it work on ARM64 but I haven't personally tried.

joperezr, 2019-09-10:
As @Frankenslag suggests, for most cases you have the regular Raspbian distrubuted on the raspberry pi site, which would be ARM 32. Closing this as the question has been answered. That said, feel free to reopen if you have any follow up questions.

closed 2019-09-10 by joperezr