dotnet/iot #721: Output initialization problem

aaronchch, 2019-09-15:
I have a Raspberry Pi 3B and I'm using an output to control a relay board. The relay board requires and active-low signal
Using the code below to initialize the output causes it to go low for long enough for the relay to briefly close before the output goes high. 

```
GpioController controller = new GpioController(PinNumberingScheme.Logical);
controller.OpenPin(18, PinMode.Output);
controller.Write(18, PinValue.High);
```

What is needed is a way of setting the output state of the pin before configuring it as an output. Calling GpioController.Write before opening the pin throws an exception.

Frankenslag, 2019-09-16:
@aaronchch 

How about

```
GpioController controller = new GpioController(PinNumberingScheme.Logical);
controller.OpenPin(18);
controller.Write(18, PinValue.High);
controller.SetPin Mode(18, PinPinMode.Output);
```

Or perhaps....

```
GpioController controller = new GpioController(PinNumberingScheme.Logical);
controller.OpenPin(18, PinMode.InputPullUp);
controller.Write(18, PinValue.High);
controller.SetPin Mode(18, PinPinMode.Output);
```

Note that I haven't tried any of this and I don't know if it is safe to do or even if it will work...

krwq, 2019-09-16:
@aaronchch interesting scenario - I think we should add this API assuming this is supported by the OS. For now for Raspberry PI on Linux you can also access registers directly using:
https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/Drivers/RaspberryPi3Driver.Linux.cs#L313

note they are protected so you will need to inherit from that class, expose those 2 APIs and pass in the instance to GpioController.

What about something like: `controller.OpenPin(18, PinMode.Output, PinValue.High;`

joperezr, 2019-09-16:
I think that even if we get the initial value on the OpenPin method, we might have problems honoring this in some drivers (like the unix driver) but we can try do our best to set it just as the pin gets opened. I have triaged this in for our vNext as I think it would make sense to add support for this.

krwq, 2019-09-16:
@joperezr I think we should just document API as "best effort but no guarantee" (similarly as CancellationToken for Async APIs) - we can also add full list of drivers which would support this

aaronchch, 2019-09-17:
Sounds perfect and thanks for the workaround.

vicente-da-silva, 2021-01-10:
Any feedback on when this will be implemented? Having the relays bounce on application startup is not good.

pgrawehr, 2021-01-10:
There's a PR open to potentially fix this: #1361. We'll be discussing this on our next meeting.

krwq, 2021-01-14:
Triage: API `controller.OpenPin(18, PinMode.Output, PinValue.High);` is approved as long as we find at least one driver which supports it. And we should fallback to the Open+Write behavior if it's not possible to achieve with this driver. If we can't find any driver which can do this as atomic operation we should not add this API.

PR #1473 (2021-05-20): A GPIO has to be writeable even in drive mode input because one wants to set the value used when switching to output mode.
Fixes #721 
Fixes https://github.com/dotnet/iot/issues/1222
Superseedes #1361 

Adds two new methods to GpioController:
```csharp
OpenPin(int pinNumber, PinMode mode, PinValue initialValue);
SetPinMode(int pinNumber, PinMode mode, PinValue initialValue);
```

Writing a value before changing the mode, so that the pin immediatelly enters that state (or stays where it is), seems to work just fine on the Raspi. The other boards will currently emulate the feature as good as pos [... cut]
closed 2021-05-20 by msftbot[bot]