dotnet/iot #725: GSM, GPRS, LTE AT commands

krwq, 2019-09-18:
One of the common IoT scenarios is to:
- send/receive text message
- answer/make a phone call

We should add support for AT commands - if there is anything device specific we should create also couple of device bindings if not then just create one generic device for that.

shaggygi, 2019-09-18:
Found this...

https://www.telit.com/wp-content/uploads/2018/01/Telit-LE910-V2-Modules-AT-Commands-Reference-Guide-r3.pdf
-

krwq, 2023-02-02:
[Triage] We're not tracking the device bindings requests. If anyone is interested in adding support please send a PR

krwq, 2023-02-02:
There are existing libraries like i.e. https://github.com/hbjorgo/ATLib which can do that.

closed 2023-02-02 by krwq