dotnet/iot #727: Instructions for adding package from README.md don't work

krwq, 2019-09-19:
SDK: `3.0.100-rc1-014190`

```
mkdir test
cd test
export DOTNET_MULTILEVEL_LOOKUP=0
dotnet new console
dotnet add package System.Device.Gpio --source https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json
```

Output:

```
  Writing /tmp/tmpF4NDO5.tmp
info : Adding PackageReference for package 'System.Device.Gpio' into project '/tmp/myapp/myapp.csproj'.
info : Restoring packages for /tmp/myapp.csproj...
info :   CACHE https://dotnetfeed.blob.core.windows.net/dotnet-iot/flatcontainer/system.device.gpio/index.json
info :   CACHE https://api.nuget.org/v3-flatcontainer/system.device.gpio/index.json
error: Unable to find a stable package System.Device.Gpio with version 
error:   - Found 200 version(s) in https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json [ Nearest version: 1.1.0-prerelease.19467.2 ]
error:   - Found 6 version(s) in nuget.org [ Nearest version: 1.0.0-prerelease.19413.1 ]
error: Package 'System.Device.Gpio' is incompatible with 'all' frameworks in project '/tmp/myapp/myapp.csproj'.
```

When trying the "nearest version":

```
dotnet add package System.Device.Gpio --source https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json --version 1.1.0-prerelease.19467.2
```

it works correctly for both packages.

This should work by default or instructions should be updated

joperezr, 2019-09-19:
I believe this is because you are not passing in the --Prerelease flag so it is only looking for stable versions. Can you try that to see if it fixes it? If so, we can just add that to the instructions (although they will have to change anyway very soon once we ship since we won't need the flag nor the extra --source parameters.)

krwq, 2019-09-19:
@joperezr I couldn't see it in the help (`dotnet add package --help`) - perhaps we should file an issue on dotnet?

joperezr, 2019-09-19:
we could do that yes, but did it actually fix the issue?

krwq, 2019-09-19:
@joperezr haven't had chance to try yet - will let you know when I have a chance.

Also this will presumably fix itself on Monday anyway, right? (will try to test before)

Ellerbach, 2020-12-22:
@krwq do we still have this issue or we can close it?

krwq, 2020-12-22:
I don't think this is working `--prerelease` is downloading 1.2.0 for me rather than nightly

krwq, 2023-02-02:
[Triage] We don't think it's an issue anymore

closed 2023-02-02 by krwq