dotnet/iot #733: About addressable RGB LEDs

HumJ0218, 2019-09-20:
Now here we have WS28xx. But it seems WS28xx doesn't fit perfect.
I have completed a driver for [SK9822](https://github.com/HumJ0218/HumphreyJ.NetCore.Devices/tree/master/HumphreyJ.NetCore.Devices/SPI/SK9822). The SK9822 has a clock line so it can be controlled very stablely.
And APA102, it's controlled exactly the same way as SK9822.

These all are addressable RGB LEDs. Should they all be extended from the same class?

krwq, 2019-09-20:
@HumJ0218 we currently do base classes only if implementation is relatively similar. If they differ you might want to change separate device for now (ideally with similar APIs).
For now refrain from creating any abstractions

fkromer, 2021-10-18:
@HumJ0218 Could you add a license to your project? Relates to https://github.com/dotnet/iot/issues/1696.

krwq, 2025-02-06:
[Triage] Closing since there was no attention since 2021 and we prefer PR directly for device improvements.

closed 2025-02-06 by krwq