dotnet/iot #736: adding to c# library (dll) fails because of dependecies

sheepchen, 2019-09-22:
If i want to add the nuget-package to a c# library (dll), it fails because of the dependecy of "-NetStandard,version=v2.0". C#-librarys are Net-Framework (4.72)

"Error. The package "System.Runtime.WindowsRuntime 4.3.0"can't be installed. You try to install the pakage in a project that is based on ".NETFramework,Version=v4.7.2". The paket has no Assemblylinks that a compatible. "

**Steps to reproduce**
Create a C# library and try to add System.Device.Gpio via nuget-pakage-manager

**Expected behavior**
installing works.

**Actual behavior**
installing didn't work.

**Versions used**
Visual Studio 2019
Framework 4.72 for the dll
dotnet 2 for the binary that uses the dll

- Version of `System.Device.Gpio` package
tried any version

joperezr, 2019-09-24:
Hello @sheepchen can you please share a sample repro of the project where you are seeing the failure? I tried this locally on a simple net472 class library and a netstandard2.0 library and the package is added successfully. How are you adding the package dependency?

Also, if your library is .NET Framework 4.7.2, where are you planning your library to run? I'm only curious since most of our use cases are people using .NET Core in order to be able to run on hardware like Raspberry Pis

sheepchen, 2019-09-25:
Hey,
thanks for replying. 

I tested around a little bit and have create a testprogram and found my problem.

When I say I want a C# library, VS2019 asumes it should be NetFramework. I have to specify that it should be netCore at creating. changing afterwarts is not possible.

So I have to create a new project and copy the code.

The netframework stuff from above was because I have a really big self-written library for all my projects. I started it ages ago where there were no dotnet.

Thanks.

closed 2019-09-25 by sheepchen