dotnet/iot #745: Does this library support Intel Atom series chipset like Apollo Lake?

petercoin, 2019-09-26:
I would like to know does this library support Apollo Lake, Braswell or other Atom series chipset?
From OEM/ODM viewpoint, development board might not be suitable for their business model.

krwq, 2019-09-26:
@petercoin I haven't tried on this specific setup but I think most of the code should work on any linux x64,x86,arm,arm64 (arm v7 or higher)

petercoin, 2019-09-26:
@krwq The key point is that for industrial  embedded motherboard based on x86 architecture, it is difficult to access GPIO, I2C, SPI if you don't have specific I/O driver like winio. Therefore, it will be very useful if this library supports the access to above I/Os. I will try it, thank you very much.

krwq, 2019-09-26:
@petercoin I'd try installing libgpiod which should work on any recent kernel and works relatively well. We already have a driver for it.

For Windows I think you'd likely need to write a GpioDriver

petercoin, 2019-09-27:
@krwq Intel does provide GPIO and I2C drivers but I am not sure if .Net Core IoT library can access GPIO and I2C through these drivers. However, I will find an Apollo Lake system to confirm it. 
Thank you for your reply and I will close this issue after confirming this library works fine.

petercoin, 2019-10-01:
@krwq I have tried it on Apollo Lake with Windows 10 1803 and it works fine. The most important thing is rhproxy. I think the mechanism is the same with the namespace: Windows.Device.Gpio, is it right? 
BTW, I think it is better to add a note about rhproxy for user who wants to access I/Os by this library with Windows.

closed 2019-10-01 by petercoin

krwq, 2019-10-01:
@petercoin I'm not sure I understand. If you're using Windows.Device.Gpio then you're not using System.Device.Gpio (System.Device.Gpio is using Windows.Device.Gpio underneath on Windows). If you're talking about APIs being similar then yes they were inspired by both Windows and Linux APIs

petercoin, 2019-10-02:
@krwq Sorry for the confusion. I tried both Windows.Devices.Gpio on UWP and System.Devices.Gpio on .Net Core and both of them work fine but rhproxy enabled is needed. I just want to confirm if the mechanism is similar between Windows.Devices.Gpio and System.Devices.Gpio on Windows.