dotnet/iot #748: DHT11 not working

tshaug, 2019-09-29:
**Describe the bug**

Hi everybody,

I installed the latest version of system.Device.GIO (v.1.0.0) and Iot.Device.Bindings (1.0.0)
Reading the DHT11 fails using the mentioned assemblies and .NET core 2.2 and 3.0

I used two different breakout boards (have DHT11 each). I have been using them with WIndows IOT so they could be "ok" (regarding the wiring).

I always receive no temp and humidity and IsLastReadSuccessful is always false.
I randomly receive results, like this (5 seconds delay between each measurement):
Guiding Proxy Console 1.2
Temp:  NaN
Humi:  NaN
ok?:  False

Temp:  NaN
Humi:  NaN
ok?:  False

Temp: 23
Humi: 44
ok?: True

Temp: 23
Humi: 44
ok?: True

Temp: 23
Humi: 44
ok?: True 

Temp: 23
Humi: 44
ok?: True

Temp: 23
Humi: 44
ok?: True

Temp: NaN
Humi: $NaN
ok?: False

Temp: 23
Humi: 44
ok?: True

Temp: NaN
Humi: NaN
ok?: False

Any ideas?

Thank you very much!
Thomas

**Steps to reproduce**

```csharp
  using (Dht11 dht = new Dht11((4), PinNumberingScheme.Logical))
            {
                for (int i = 0; i < 3; i++)
                {
                    Temperature temperature = dht.Temperature;
                    double humidity = dht.Humidity;

                    Console.WriteLine($"Temp: {temperature.Celsius}");
                    Console.WriteLine($"Humi: {humidity}");
                    Console.WriteLine($"ok?: {dht.IsLastReadSuccessful}");
                    Thread.Sleep(5000);

                }
            }
```

**Expected behavior**

print temp and humidity

**Actual behavior**

Temp: NaN
Humi: NaN

(same behaviour for .NET Core 2.2 and 3.0)

**Versions used**

Add following information:

- dotnet on rasppi 3:
2.2 and 3.0

2.2 
   .NET Core SDK (reflecting any global.json):
   Version:   2.2.102
   Commit:    96ff75a873

  Runtime Environment:
      OS Name:     raspbian
      OS Version:  9
      OS Platform: Linux
      RID:         linux-arm
      Base Path:   /home/pi/dotnet/sdk/2.2.102/

     Host (useful for support):
        Version: 2.2.1
        Commit:  878dd11e62

        .NET Core SDKs installed:
         2.2.102 [/home/pi/dotnet/sdk]

       .NET Core runtimes installed:
             Microsoft.AspNetCore.All 2.2.1 [/home/pi/dotnet/shared/Microsoft.AspNetCore.All]
             Microsoft.AspNetCore.App 2.2.1 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
            Microsoft.NETCore.App 2.2.1 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

.NET Core 3.0 
   .NET Core SDK (reflecting any global.json):
    Version:   3.0.100-rc1-014190
    Commit:    c4d43f672d

    Runtime Environment:
         OS Name:     raspbian
         OS Version:  9
         OS Platform: Linux
         RID:         linux-arm
         Base Path:   /home/pi/dotnet3/sdk/3.0.100-rc1-014190/

         Host (useful for support):
            Version: 3.0.0-rc1-19456-20
            Commit:  8f5d7b1ba4

          .NET Core SDKs installed:
              3.0.100-rc1-014190 [/home/pi/dotnet3/sdk]

           .NET Core runtimes installed:
                Microsoft.AspNetCore.App 3.0.0-rc1.19457.4 
                [/home/pi/dotnet3/shared/Microsoft.AspNetCore.App]
                 Microsoft.NETCore.App 3.0.0-rc1-19456-20 [/home/pi/dotnet3/shared/Microsoft.NETCore.App]

Build machine:
      .NET Core SDK (reflecting any global.json):
       Version:   3.0.100
       Commit:    04339c3a26

      Runtime Environment:
            OS Name:     Windows
            OS Version:  10.0.17763
            OS Platform: Windows
            RID:         win10-x64
           Base Path:   C:\Program Files\dotnet\sdk\3.0.100\

        Host (useful for support):
          Version: 3.0.0
          Commit:  7d57652f33

           .NET Core SDKs installed:
               2.1.701 [C:\Program Files\dotnet\sdk]
               2.1.801 [C:\Program Files\dotnet\sdk]
               2.1.802 [C:\Program Files\dotnet\sdk]
               2.2.301 [C:\Program Files\dotnet\sdk]
               2.2.401 [C:\Program Files\dotnet\sdk]
               2.2.402 [C:\Program Files\dotnet\sdk]
               3.0.100 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

- Version of `System.Device.Gpio` package = 1.0.0
- Version of `Iot.Device.Bindings` package = 1.0.0

krwq, 2019-09-30:
Yes, 1-wire currently does not work too well. I believe we will need to fix this first though: https://github.com/dotnet/iot/issues/197 and then update DHT code

tshaug, 2019-09-30:
Good morning Krzysztof,

thank you for your feedback. 
The referenced item [197](https://github.com/dotnet/iot/issues/197) is quite old. 
is there already somne rough idea when an updated 1.wire implementartion will be released (even i beta) I am happy to test ;-)

Cheers
Thomas

krwq, 2019-09-30:
@tshaug it's currently on [our roadmap for the next release](https://github.com/dotnet/iot/blob/master/Documentation/roadmap.md) but we prioritize based on demand so definitely this issue will bump the priority a bit.

1-wire in general is quite not reliable (it's timing based and we're not working under real-time OS so any thread/context switch at a bad time can destroy the reading process).

Having said that there are two ways we can go about improving the current state of DHT11:

- improve (current) bit banging implementation - currently it works around 10% of the time which I bet we can improve
- add proper implementation like mentioned in #197 which is based on the kernel driver and later convert bit-banging implementation to SoftwareOneWire (similar done as we've done for SpiDevice and SoftwareSpiDevice) which will allow people to swap based on demand.

The fastest way to get it would probably be to contribute, otherwise we are talking about around 1 year timeline with a `maybe` factor (even though we want we we might not have time to focus on this).

I believe in general using kernel driver implementation will not be super hard and I can help with that - all you need to do is to read and parse couple of files from under `/dev/` as per pointers in the other issue.

Alternative and simpler solution might be to change thermometer to anything not 1-wire based (I2C/SPI/analog). You can see supported thermometers here: https://github.com/dotnet/iot/tree/master/src/devices#thermometers

tshaug, 2019-10-01:
Good morning Krzysztof

thank you for your explanation and the alternatives. 
I assume that I will go for an I2C based sensor (I would like to contribute, but I have not programmed in C for  almost 20 years, so I would assume that it will take me to long to get to speed...)". 
I need a sensor for humidity so the list of thermometers provide some alternatives - thanks 

Cheers
Thomas

krwq, 2019-10-01:
@tshaug for humidity sensors see:
https://github.com/dotnet/iot/tree/master/src/devices#hygrometers

From my personal preference I'd take something from https://github.com/dotnet/iot/blob/master/src/devices/Bmxx80/README.md family (i.e. BME280) but if you just need humidity it might be better to get something cheaper

tshaug, 2019-10-01:
Hi Krzysztof

thank you once again for yur advice. 
I have already order 3 BME sensor I have alsoray exerience using BMP180 and BMP 280, but I did't know that the BME280 is also providing humidity. I found that based on your previous advice. 
I hope to receive them tomorrow then I can start to "integrate" ;-)

Thanks again
Thomas

krwq, 2019-10-01:
@tshaug cool to hear. Please also make sure to look at https://github.com/dotnet/iot/issues/753 and https://github.com/dotnet/iot/issues/752

If you are only reading humidity/temperature then it won't affect you - for pressure the first issue mentions that you also need to change the power mode before using it - otherwise you'll get incorrect pressure reading.

The issues will likely result in some usability improvements and therefore some APIs will change shortly (there won't be anything drastic but if you happen to upgrade Iot.Device.Bindings package soon you might notice that something works a bit differently).

If you just copy & paste the sample code then you don't need to change anything

joperezr, 2019-10-10:
@krwq do you want to close this one as a dupe of the other existing issues of DHT?

krwq, 2019-10-10:
Yes, I think there is no new work this issue introduces from other issues although this one describes user facing problem a bit clearer

closed 2019-10-10 by krwq