dotnet/iot #749: Is Raspberry PI supported?

alvipeo, 2019-09-29:
Does .Net Core 3.0 support Raspberry Pi? If so, which versions?

Can't find this anywhere!

krwq, 2019-09-30:
Raspberry Pi 2, 3 and 4 do work. Raspberry Pi Zero does not because it's ARMv6 and .NET Core works only on ARMv7 or higher.

shaggygi, 2019-09-30:
https://github.com/dotnet/iot/tree/master/samples#requirements

alvipeo, 2019-09-30:
That's good news! Thank you!

Another question - can I work with Pi's camera module with .Net Core 3.0?

krwq, 2019-09-30:
@alvipeo .NET Core 3.0 + nightly build of dotnet/iot can do still images (Videos will need additional work):
I've literally just merged support for that in https://github.com/dotnet/iot/pull/664 - packages should be available any time

alvipeo, 2019-09-30:
ooooh!!! you're my hero! :)
any ETA?

krwq, 2019-09-30:
Thank @ZhangGaoxing since it was him who added the support :smile:

@joperezr will https://github.com/dotnet/iot#net-cli use nightly or stable?

Note that this repo is slightly independent from .NET Core - we do produce NuGet packages at the same time as .NET Core but you can upgrade IoT independently to .NET Core

alvipeo, 2019-09-30:
ok, please just tell me how do I know when camera support is released?

joperezr, 2019-09-30:
@alvipeo do you require the package to be stable in order for you to consume it? If not you can just use our preview builds by just adding our nuget feed to your restore sources: https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json. I have just queued a new build with the changes you require, and will let you know which version to depend on which will have this support.

krwq, 2019-09-30:
If you require stable then per https://github.com/dotnet/core/blob/master/roadmap.md#upcoming-ship-dates

closest  release is 5.0 which ships Nov 2020.

joperezr, 2019-09-30:
Build is done. You should be able to consume package version `1.1.0-prerelease.19480.2` from the feed I specified above. This version will have the support you require.

alvipeo, 2019-09-30:
@joperezr I don't know. I require that it just works ;)

ok, thanks a lot! I'll try it!

EdiWang, 2019-10-02:
Check my blog post 'Setup .NET Core 3.0 Runtime and SDK on Raspberry Pi 4' for a complete 
guide, you can eventually get a ASP.NET Core 3.0 web server running on your Pi. https://edi.wang/post/2019/9/29/setup-net-core-30-runtime-and-sdk-on-raspberry-pi-4

alvipeo, 2019-10-02:
Thank you!  I will!

joperezr, 2019-10-02:
Given the question above has been answered, I will go ahead and close this issue for now. @alvipeo feel free to re-open the issue if you have any follow up questions.

closed 2019-10-02 by joperezr