dotnet/iot #750: Support 7-Segment modules on Max7219 Implementation

cerkit, 2019-09-30:
The current implementation for the Max7219 device is aimed directly at the LED Matrix. However, the Max7219 is also used to drive 7-Segment displays. A few of the constants force the Digit count to be 8. There are use-cases where an 8 digit 7-Segment display module may only use a few of the digits (not all 8).

Here is a link to ebay showing [some of the devices](https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=max7219+7-segment&_sacat=0) that I'm referring to.

It looks like the current Max7219 class could be used with a 7-segment display with some modifications to the Init() method, but there is still some matrix-specific code in the implementation.

[Max7219 7-Segment Display (image)](https://cerkit.com/content/images/2017/11/7-segment-bad-food--2-.jpg)

Here is a list of the byte values for displaying numbers/characters on the device:

{'0', 0x7e},
{'1', 0x30},
{'2', 0x6d},
{'3', 0x79},
{'4', 0x33},
{'5', 0x5b},
{'6', 0x5f},
{'7', 0x70},
{'8', 0x7f},
{'9', 0x7b},
{'A', 0x77},
{'B', 0x1f},
{'C', 0x4e},
{'D', 0x3d},
{'E', 0x4f},
{'F', 0x47},
{'O', 0x1d},
{'R', 0x05},
{'-', 0x40},
{' ', 0x00}

(This allows you to display the word "Error" as well as all 16 Hex digits and a space (blank)). This is taken from my hobby [blog post](https://cerkit.com/2017/11/26/driving-the-max-7219-7-segment-display-module-from-asp-net-core-on-the-raspberry-pi/) on the topic.

joperezr, 2019-10-10:
cc: @schonbachler who originally wrote this binding.

Hey @cerkit that seems reasonable. In general, we don't require full implementations of bindings in order to merge them into the repo, since some support for some scenarios (as long as they are valid/reasonable scenarios) is better than none. Would you be interested in adding this extra functionality to the existing binding?

krwq, 2023-02-02:
[Triage] We're not tracking the device bindings requests (and this kinda falls into this category). If anyone is interested in adding support please send a PR

closed 2023-02-02 by krwq