dotnet/iot #751: [Question] Is OrangePi supported?

alexk8, 2019-09-30:
There is a device family based on different arm64 chips (allwinner H3,H5, etc).
They can run on Armbian linux. Also I managed to run net core successfully few months ago.
Are their GPIO supported in anyway?

ZhangGaoxing, 2019-10-01:
I'm currently trying to add full support(pull-up, pull-down) for Allwinner GPIO through control registers. 
The code is almost complete, and after I have cleaned up the code and done more testing, I can submit a PR. @krwq

krwq, 2019-10-01:
@ZhangGaoxing awesome! @joperezr any opinion on putting those under System.Device.Gpio? I've mixed feeling because it feels like that's better place but at the same time System.Device.Gpio has CI coverage and this likely wouldn't have coverage any time soon.

joperezr, 2019-10-10:
Agreed, I think it would be good to introduce this for now as part of the Iot.Device.Bindings package, and once we are ready to add full support (and CI coverage) for it then we can move it to the main library.

closed 2023-02-02 by krwq