dotnet/iot #752: Generic calculations taken from temperature, relative humidity and pressure

MarkCiliaVincenti, 2019-09-30:
There are some formulas that can be done from sensor data, such as taking the temperature, relative humidity and pressure and outputting the absolute humidity.

Or getting the dew point from temperature and relative humidity.

Also a constant for the default sea level pressure, which I believe is incorrect in Bme280.sample.cs (I believe it should read 1013.25 instead of 1033.00).

Where would be the best place to put these? I'd like to make a pull request.

krwq, 2019-09-30:
@MarkCiliaVincenti you will need to inspect:
https://github.com/dotnet/iot/tree/master/src/devices/Bmxx80

if the calculation/data is BME280 specific (or you don't have knowledge/devices to test) then put it directly under Bme280.cs otherwise upgrade on the base class.

For sea level pressure - I'd recommend creating a separate PR for that since this might trigger some extra conversations. I agree that **mean** sea level pressure should be 1013.25  but **mean** implies that it differs depending on location (if I was living at the sea level and it always read 1033 I'd strongly argue with people claiming 1013.25).
I can currently see only using that value in the samples though - which specific API are you referring to? (if this is just sample then fine with updating in the same PR)

MarkCiliaVincenti, 2019-09-30:
I'm not talking about BME280-specific calculations but very generic weather formulas :)

Agreed with mean sea level pressure varying, but since 1013.25 is widely considered the global average, shouldn't this meaningful number be put somewhere in the code?

krwq, 2019-09-30:
@MarkCiliaVincenti in case of SeaLevel I think we should consider adding another unit to:
https://github.com/dotnet/iot/tree/master/src/devices/Units

and add it as a constant (or static readonly field).

Since I was already thinking of proposing Pressure unit (because seems like there are at least 4 different units of pressure) I think we should just go ahead and do it.

Full scope of this feature might require a bit more work to convert all usages of Pressure in the repo but we could help with that.

For the formula since I don't know much about this specific domain my gut feeling tells me to create new project under `devices/` called `WeatherHelpers` or some even more generic names where we could gather all helper functions (project name doesn't matter that much since we don't use it anywhere but namespace and class name does)

krwq, 2019-09-30:
Interestingly when I calculate altitude with 1013.25hPa it tells me my altitude is negative where 1033 shows 150m which sounds reasonable

krwq, 2019-10-01:
@MarkCiliaVincenti please also see https://github.com/dotnet/iot/issues/753 if there is no overlap. Also might be worth to check with @RobinTTY so that you don't get really bad merge conflict

RobinTTY, 2019-10-01:
Just adding my thoughts: I think @krwq 's idea of creating a new project with WeatherHelpers is good since the `GetAltitude()` method in the Bmx280 binding is not a sensor specific feature and instead just a calculation made from the pressure value, so it should probably haven't been in the device binding in the first place.

MarkCiliaVincenti, 2019-10-07:
Once #768 gets merged into master, I will work on the WeatherHelper.

MarkCiliaVincenti, 2019-10-08:
Pressure part has been handled by #768 which was merged to master. I'll create a new pull request for the WeatherHelper.

krwq, 2019-10-08:
@MarkCiliaVincenti cool, if possible please review all similar sensors if they don't have some other helpful utils we can re-use.

I'd go with creating library like "Utilities" and put WeatherHelper there but WeatherHelpers as a project is fine too for now but I suspect it will be too small to stay as a seperate project on its own

MarkCiliaVincenti, 2019-10-12:
This is completed by pull requests #768 and #783

MarkCiliaVincenti, 2020-04-23:
#783 was merged as well. Closing issue.

closed 2020-04-23 by MarkCiliaVincenti

krwq, 2020-04-23:
Thanks @MarkCiliaVincenti and @pgrawehr for helping to push this!