dotnet/iot #757: Linux PWM implementation throws during construction/setting frequency in some cases

krwq, 2019-10-01:
Repro:

```csharp
using (var pwm = PwmChannel.Create(0, 0, 500))
{
   pwm.Frequency = 10000;
}
```

from now constructing pwm channel with frequency 10k Hz is not possible without manually editing duty cycle.

This is happening because duty cycle still has the old value in the driver (measured in nanosecond) and updating frequency causes that duty cycle gets out of range

shaggygi, 2019-10-01:
I noticed this when we were updating to the new PwmChannel.  I believe it is because there is not enough time delayed when [opening the port (via file)](https://github.com/dotnet/iot/blob/90f0786d82afbf12357868733ce5ddb2b2b8c8e8/src/System.Device.Gpio/System/Device/Pwm/Channels/UnixPwmChannel.Linux.cs#L170) before interacting with the DC/freq.  I mentioned it then, but we couldn't figure out a good approach on timing.

I could duplicate using the [PwmPinOutput command](https://github.com/dotnet/iot/blob/master/tools/DevicesApiTester/Commands/Pwm/PwmPinOutput.cs) in the DeviceTesterApi.

After the exception occurs, you can still see the channel opened by reading the file.  And writing the DC/freq works by then, as well.

krwq, 2019-10-02:
@shaggygi, I've added a fix in https://github.com/dotnet/iot/pull/758 - I got 100% repro before and now doesn't repro with the fix.

In some scenarios this is not particularly obvious:
- you create PWM with one frequency and dispose
- you create PWM with different frequency and it throws when constructing

The issue is with duty cycle being out of range (also IO exception suggests that: `Invalid argument`)

joperezr, 2019-10-10:
Would we want to service 1.0 with this?

krwq, 2019-10-10:
I don't have strong opinion either way. On one hand this is relatively rare but on the other it's super annoying when it affects you

closed 2019-12-04 by krwq