dotnet/iot #760: BMxx80 samples are broken

ygoe, 2019-10-02:
**Describe the bug**

The sample code for the BMxx80 chips is broken. It does not compile and I cannot recover it. It needs to be updated to match the current API.

krwq, 2019-10-02:
@ygoe are you referring to code in the README.md or code in the .cs file? Code in .cs file should be up to date. README.md we can fix as part of https://github.com/dotnet/iot/issues/753

ygoe, 2019-10-03:
I couldn't find it anymore so I googled it and I'm not sure I landed on the same page. If I did, it must have been this one:

https://github.com/dotnet/iot/blob/master/src/devices/Bmxx80/samples/Bmp280.sample.cs

I've deleted my playground project again as I lost interest in it.

krwq, 2019-10-03:
@ygoe depending on the sensor you used you need to use the right sample (i.e. BMP280 sample won't work with BME280 and vice-versa)

ygoe, 2019-10-03:
I have a BMP280 sensor. It's connected to a Pi and works there in Node-RED. Then I wanted to try it out in .NET.

krwq, 2019-10-03:
@ygoe would be useful to see what kind of error you're seeing and how your project looks like. I believe @joperezr tried BMP280 around a week ago without any issues

MarkCiliaVincenti, 2019-10-07:
I updated the Bmp280.sample.cs in a pull request 2 days ago (#768) and the code compiled on OSX, Linux and Windows.

joperezr, 2019-10-10:
I have verified that the code works fine for me again when connecting a BMP280 sensor. @ygoe did you by any chance try again? Can you see your device if you run `i2cdetect 1` from the pi's terminal?

krwq, 2019-10-10:
@ygoe one thing I've frequently seen people forget/not know is that they didn't enable I2C (use raspi-config for that).

Please let us know if you were able to fix the issue.

MarkCiliaVincenti, 2019-10-14:
> @ygoe one thing I've frequently seen people forget/not know is that they didn't enable I2C (use raspi-config for that).
> 
> Please let us know if you were able to fix the issue.

So why don't we make that clear by providing some inline comments warning the reader to enable I2C, within the sample CS files?

krwq, 2019-10-15:
@MarkCiliaVincenti I'd rather have a single place we document how to work with I2C and link this doc whenever we remember, otherwise we're gonna end up repeating the same message on all devices. Despite after a while working with this repo this seems kind of obvious and it's starts being as obvious as "plug in your device first" which I don't believe we should document anywhere...

It's one of the hundred things someone has to do and I only have limited amount of time per day :-)

MarkCiliaVincenti, 2019-10-15:
> @MarkCiliaVincenti I'd rather have a single place we document how to work with I2C and link this doc whenever we remember, otherwise we're gonna end up repeating the same message on all devices. 

@krwq this is exactly what I meant, that's why I recommended putting in a URL with the instructions.

> Despite after a while working with this repo this seems kind of obvious and it's starts being as obvious as "plug in your device first" which I don't believe we should document anywhere...

Not really a fair comparison. One is obvious that you need to plug in the device, the other is not intuitive, especially to software developers like me who are trying to work on projects while lacking any electrical engineering background.

> It's one of the hundred things someone has to do and I only have limited amount of time per day :-)

Well, there's that too which I can sympathize with 💯

krwq, 2019-10-15:
@MarkCiliaVincenti I also found it not obvious that I2C needs to be enabled at first after a few months working with the repo it became natural knowledge. I agree there should be some kind of guideliness

krwq, 2019-10-15:
I'm closing this issue since there is no response here and we have couple of confirmation about samples. @ygoe let us know details on what kind of issue you have hit if you have a chance to try again and re-open if you feel like there is anything missing.

I've created a separate issue for doc updates: https://github.com/dotnet/iot/issues/802

ygoe, 2019-10-16:
Back from holidays and worked my way through my inbox. Thanks for the comments first. As I already explained in my first comment, the code did not compile, and also that the sensor works perfectly in Node-RED on the same Raspberry Pi. So any comments about enabling I2C are irrelevant for two reasons.

I just tried it again. Here's what I did:

* Open VS 2019, v16.3.2
* Create a new .NET Core console project
* Replace the default main code with the complete file as [linked above](https://github.com/dotnet/iot/blob/master/src/devices/Bmxx80/samples/Bmp280.sample.cs)
* Install the 2 NuGet packages as explained in the [repo's README file](https://github.com/dotnet/iot): System.Device.Gpio and Iot.Device.Bindings

I have 16 compiler errors now. Here's the full listing:

```
Schweregrad	Code	Beschreibung	Projekt	Datei	Zeile	Unterdrückungszustand
Fehler	CS0138	Eine "using-Namespace"-Anweisung kann nur auf Namespaces angewendet werden. "FilteringMode" ist ein Typ und kein Namespace. Verwenden Sie stattdessen eine "using static"-Anweisung	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	9	Aktiv
Fehler	CS0103	Der Name "Pressure" ist im aktuellen Kontext nicht vorhanden.	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	24	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TemperatureSampling", und es konnte keine zugängliche TemperatureSampling-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	35	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "PressureSampling", und es konnte keine zugängliche PressureSampling-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	36	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "GetMeasurementDuration", und es konnte keine zugängliche GetMeasurementDuration-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	42	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadTemperature", und es konnte keine zugängliche TryReadTemperature-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	46	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadPressure", und es konnte keine zugängliche TryReadPressure-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	48	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadAltitude", und es konnte keine zugängliche TryReadAltitude-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	50	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TemperatureSampling", und es konnte keine zugängliche TemperatureSampling-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	55	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "PressureSampling", und es konnte keine zugängliche PressureSampling-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	56	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "FilterMode", und es konnte keine zugängliche FilterMode-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	57	Aktiv
Fehler	CS0103	Der Name "Bmx280FilteringMode" ist im aktuellen Kontext nicht vorhanden.	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	57	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "GetMeasurementDuration", und es konnte keine zugängliche GetMeasurementDuration-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	63	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadTemperature", und es konnte keine zugängliche TryReadTemperature-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	67	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadPressure", und es konnte keine zugängliche TryReadPressure-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	69	Aktiv
Fehler	CS1061	"Bmp280" enthält keine Definition für "TryReadAltitude", und es konnte keine zugängliche TryReadAltitude-Erweiterungsmethode gefunden werden, die ein erstes Argument vom Typ "Bmp280" akzeptiert (möglicherweise fehlt eine using-Direktive oder ein Assemblyverweis).	PiNet	C:\Source\Experiments\PiNet\PiNet\Program.cs	71	Aktiv
```

(Sorry, that's what I get when I copy stuff from the errors panel.)

What have I done wrong?

krwq, 2019-10-17:
@ygoe can you try removing package references (edit csproj) and add them again but use `1.1.0-prerelease.19517.1` of the package? (i.e. `dotnet add package Iot.Device.Bindings --version 1.1.0-prerelease.19517.1`)

jmork, 2019-11-13:
The problem is that the library seems to have been rewritten to use async, but the samples have not been updated accordingly

joperezr, 2019-11-13:
Reopening to track the fix of the samples to  use async instead.

Ellerbach, 2020-12-22:
All samples had been updated.

closed 2020-12-22 by Ellerbach