dotnet/iot #762: Support for GPIO Bank 1 when using Raspberry Pi Compute Module 3

mjjames, 2019-10-03:
Although the Raspberry PI 3 Driver works very well it is currently limited to the number of GPIO pins on the Raspberry PI 3. The Raspberry Pi Compute Module 3 has an additional bank of GPIO's so you can use 1-45.

I propose that we make the [ValidatePinNumber method protected virtual](https://github.com/dotnet/iot/blob/2399a1dfd3020934d9939b3a27ace953bc39235a/src/System.Device.Gpio/System/Device/Gpio/Drivers/RaspberryPi3Driver.cs#L17) so that a RaspberryPiComputeModule3 driver is introducted that inherits from the RaspberryPi3 Driver and simply extends the known GPIO's.

Everything else is pretty much the same in terms of implementation I believe.

mjjames, 2019-10-03:
I've created a form and started working on this. The `GpioController` also needs updating as the `GetBestDriverForRaspberryPiRevision` method needs to consider the revision number to determine to use the CM3 driver or the current Pi 3 driver

Frankenslag, 2019-10-03:
@mjjames 
I had a quick scan of the fork and note that you are detecting the cm3 by using values read from procinfo. You may have problems with this outside of raspbian please see #661

mjjames, 2019-10-03:
I agree, happy to discuss if working on #661 is required before completing this. It makes sense to fix an underlying issue than build on top of it

Frankenslag, 2019-10-03:
@mjjames 

I have asked @joperezr if  #661 should be resurrected now. It was postponed as we didn’t want to clash with the already mostly tested 3.0 release and there is a workaround by explicitly specifying the driver.

What #661 is trying to do is identify a pi whereas your proposed pr is looking for the type of pi and I guess they are separate and the connection is that they are in the same code area and need to be aware of the way that other os’s work. This is why I chose to read the device tree as it was part of the firmware package and likely to be available on all Linux based os’s.

Thoughts are that if @joperezr agrees I resubmit the or associated with #661 as it improves on something already there and you, knowing what’s in there, can continue to work on your pr.

Very happy to discuss further.

mjjames, 2019-10-04:
After coming back to this with fresh eyes, the `ConvertPinNumberToLogicalNumberingScheme` doesn't make alot of sense for the ComputeModule3. Would it be acceptable to throw a NotSupportedException here?

Frankenslag, 2019-10-04:
@mjjames 

PR #766 created.

joperezr, 2019-10-04:
Is the problem here really just that when you try to use the existing RaspberryPi3 driver it is failing because of validation when trying to do an operation on those extra GPIOs? If so, I would rather provide a way to opt-out of that validation or to  pass in the number of pins that should be available instead of creating a whole new driver for this.

krwq, 2019-10-11:
@mjjames 

> After coming back to this with fresh eyes, the ConvertPinNumberToLogicalNumberingScheme doesn't make alot of sense for the ComputeModule3. Would it be acceptable to throw a NotSupportedException here?

Yes, fine to throw if it doesn't make sense. Also please try to make the code as simple as possible (basically what @joperezr wrote)

> @krwq can you comment on this and regarding the tests earlier up the convo?

Not sure what's the question is about the tests but adding implementation for compute module as a workaround and then improving detection logic as a next step sounds good to me.

JTrotta, 2020-03-30:
Any update about bank 1?

joperezr, 2020-03-31:
I believe @mjjames was working on it but that PR seem to have gone stale. @JTrotta would you be interested in taking @mjjames work and finishing it in case he doesn't want to work on it anymore?

JTrotta, 2020-04-01:
@joperezr ,will do.

pgrawehr, 2020-04-01:
@JTrotta Building with VS2019 works fine, with a few quircks you have to know. First thing is that you need to run the build.cmd file first, before opening any solution. What error message(s) do you get?

JTrotta, 2020-04-01:
@pgrawehr  have a look here 

> https://github.com/dotnet/iot/issues/784#issuecomment-607079901

JTrotta, 2020-04-02:
Added Pull request #1034
Tested on my CM3, Pins over 32 now work

joperezr, 2020-04-02:
I'll take a look at that PR shortly, thanks for adding it!

krwq, 2023-02-02:
[Triage] This should be already fixed by https://github.com/dotnet/iot/pull/1200 please re-open if there is anything left

closed 2023-02-02 by krwq