dotnet/iot #764: Error 2 Initializing the Gpio driver

JPThorne, 2019-10-03:
**Description**

I get the following error when attempting to use `System.Device.Gpio` and `GpioController`: 

```
Let's blink an LED!

Unhandled Exception: System.IO.IOException: Error 2 initializing the Gpio driver.
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at Athy.Device.Simulation.Program.Main(String[] args) in /app/src/Athy.Device.Simulation/Program.cs:line 18
```

**Steps to reproduce**

When running this `Program.cs` via the below `Dockerfile`:
```
using System;
using System.Device.Gpio;
using System.Threading;

namespace Athy.Device.Simulation
{
    class Program
    {
        static void Main(string[] args)
        {
            var pin = 17;
            var lightTimeInMilliseconds = 1000;
            var dimTimeInMilliseconds = 200;

            Console.WriteLine($"Let's blink an LED!");
            using (GpioController controller = new GpioController())
            {
                controller.OpenPin(pin, PinMode.Output);
                Console.WriteLine($"GPIO pin enabled for use: {pin}");

                Console.CancelKeyPress += (object sender, ConsoleCancelEventArgs eventArgs) =>
                {
                    controller.Dispose();
                };

                while (true)
                {
                    Console.WriteLine($"Light for {lightTimeInMilliseconds}ms");
                    controller.Write(pin, PinValue.High);
                    Thread.Sleep(lightTimeInMilliseconds);
                    Console.WriteLine($"Dim for {dimTimeInMilliseconds}ms");
                    controller.Write(pin, PinValue.Low);
                    Thread.Sleep(dimTimeInMilliseconds);
                }
            }
        }
    }
}
```
Dockerfile:
```
FROM mcr.microsoft.com/dotnet/core/sdk AS build

RUN dotnet --info

RUN mkdir /app
COPY . /app

WORKDIR /app
RUN dotnet restore ./Athy.Collection.Device.sln
RUN dotnet build -c Release ./Athy.Collection.Device.sln -o /build

FROM build AS publish
RUN mkdir /publish
RUN dotnet publish -c Release -o /publish ./src/Athy.Device.Simulation/Athy.Device.Simulation.csproj

# FROM mcr.microsoft.com/dotnet/core/runtime:3.0.0-preview8-buster-slim-arm64v8 as runtime
# FROM microsoft/dotnet:2.1-runtime-stretch-slim-arm32v7
FROM mcr.microsoft.com/dotnet/core/runtime:2.2.6-stretch-slim-arm32v7 as runtime
WORKDIR /app
COPY --from=publish /publish /app

ENTRYPOINT ["dotnet", "Athy.Device.Simulation.dll"]
```

**Expected behavior**

I expect no error to happen and the LED on the connected breadboard to blink. 

**Actual behavior**

The error described above happened instead. 

**Versions used**

The following versions are used: 

```
    Iot.Device.Bindings" Version="1.0.0-prerelease.19413.1"
    System.Device.Gpio" Version="1.0.0-prerelease.19413.1"
```

Add following information:

- `dotnet --info` output: 

```
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     debian
 OS Version:  10
 OS Platform: Linux
 RID:         debian.10-arm
 Base Path:   /usr/share/dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  3.0.100 [/usr/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
```

ZhangGaoxing, 2019-10-04:
You need to use the `--device` parameter to mount GPIO devices

JPThorne, 2019-10-04:
Hey @ZhangGaoxing thanks for the feedback, that sounds promising. Where would I use that parameter though? In the Dockerfile? Is there some documentation somewhere on how to use that? Thanks!

ZhangGaoxing, 2019-10-04:
try this
```
docker run --device /dev/gpiomem
```

JPThorne, 2019-10-05:
Great that worked! Thanks very much. I guess this can be closed then. Should this be documented somewhere? I couldn't see anything in the samples area: https://github.com/dotnet/iot/blob/master/samples/led-blink/README.md indicating I needed to run it with that `--device` flag.

ZhangGaoxing, 2019-10-05:
@joperezr @krwq 👆

joperezr, 2019-10-07:
Yes it should be documented, thanks a lot @ZhangGaoxing for answering the question. We will add these note to our docs for running on a docker container.

krwq, 2023-02-02:
[Triage] we need to move https://github.com/dotnet/iot/blob/main/samples/led-blink/README.md#running-in-containers to a more reusable place

krwq, 2025-02-06:
[Triage] Using https://github.com/dotnet/iot/issues/2261 to track this

closed 2025-02-06 by krwq