dotnet/iot #776: IsPinOpen() doesn't work for output pins

pawelpetruch, 2019-10-08:
When using System.Device.Gpio(1.0.0) on Raspbian Buster (RPi 2) in the following way:

```
private GpioController controller = new GpioController();
               
controller.OpenPin(pin, PinMode.Output);

```
and when I check the pin's status:
`controller.IsPinOpen(pin)`
it returns false when the pin is actually working fine. 

Input pins are showed correctly when IsPinOpen() is used.

- dotnet 3.0.100
- Version of System.Device.Gpio package 1.0.0

krwq, 2019-10-08:
Thanks for reporting @pawelpetruch! Quickly looking at the code I'm not sure how would that be possible. Can you show some small sample code? Are you sure you're not accidentally creating new instance of GpioController and disposing the old one?

pawelpetruch, 2019-10-09:
Thanks for the response @krwq!

A sample code:
```
class XXX
{
        private GpioController controller = new GpioController();
        
        public XXX(int pin)
        {
            controller.OpenPin(pin, PinMode.Output);
            if (controller.IsPinOpen(pin))
            {
                ...
            }
            else
            {
                ...
            }
        }
}
```

krwq, 2019-10-10:
@pawelpetruch thanks, I'll have this scenario tested either as part of https://github.com/dotnet/iot/pull/758 or sometime afterwards.

joperezr, 2019-10-15:
@krwq so is there really a problem here or what is it that we want to track for vNext? From the code that I see above, I can't seem to find a reason why this would be happening.

krwq, 2019-10-15:
At minimum we should add a test verifying this scenario works. If it's broken as suggested per issue we should fix it for vNext.

pgrawehr, 2019-10-17:
Uploaded a testcase of this Bug, but was unable to reproduce it.

krwq, 2019-10-17:
@pgrawehr do you still have any project where you can reproduce? Is it possible you're referencing some older version of System.Device.Gpio?

pgrawehr, 2019-10-18:
This was tested with the latest master build. So far, I was unable to reproduce the bug with anything I tried.

krwq, 2019-10-21:
@pawelpetruch can you help us write a test case for this issue? PR with test case is in here https://github.com/dotnet/iot/pull/806

pawelpetruch, 2019-10-22:
@krwq I will have a look over the weekend if that's fine.

krwq, 2019-10-23:
@pawelpetruch weekend sounds good, thank you!

krwq, 2019-11-26:
@pawelpetruch not trying to rush or anything - have you perhaps had a chance to look at this yet? This is currently not actionable on our side

closed 2022-06-10 by msftbot[bot]