dotnet/iot #777: Do not use SYS_TRIGGER reset on BNO055

Tragetaschen, 2019-10-08:
**Describe the bug**

According to Bosch, some BNO055 have soft-reset issues and only work with Power On Reset.
[Writing 0x20 to SYS_TRIGGER](https://github.com/dotnet/iot/blob/2440c17a6a1ba274b250ac5863bccae7f2ecc084/src/devices/Bno055/Bno055Sensor.cs#L129) might render it unusable.

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BNO055-power-on-reset-issues/m-p/8457/highlight/true#M948

**Steps to reproduce**
Depends on the individual part.
```csharp
new Bno055Sensor(…);
```
or
```bash
# i2cdetect 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1.
I will probe address range 0x03-0x77.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- 28 -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
# i2cset 1 0x28 0x3f 0x20
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will write to device file /dev/i2c-1, chip address 0x28, data address
0x3f, data 0x20, mode byte.
Continue? [Y/n]
# i2cdetect 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1.
I will probe address range 0x03-0x77.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

**Expected behavior**

Chip continues to work.

**Actual behavior**

Chip vanishes from I²C

krwq, 2019-10-08:
Good find @Tragetaschen! Seems like this is only happening when external crystal is not connected.

Are you by any chance interested in making a PR? I suspect we can just always change all registers to defaults instead of resetting.

Tragetaschen, 2019-10-21:
@krwq Yes

PR #814 (2019-11-04): Avoid the unreliable SYS_TRIGGER-based chip reset.
Fixes #777

I've decided to overwrite all the writable registers in page 0 with their default values according to the data sheet. This means, there's a change in behavior: Prior to this change, the calibration data would remain active between consecutive runs as long as you didn't change the temperature source (which was the trigger for the chip reset). With this change, that data is reset to default values every time you construct a new BNO055 driver.

I tested this on our custom board by reading the Gravity vector.
commit 0f2e35ea (2019-11-04): Avoid the unreliable SYS_TRIGGER-based chip reset. (#814)

* Avoid the unreliable SYS_TRIGGER-based chip reset.

Fixes #777

* Fix code style

* Private functions start with a capital letter
closed 2019-11-04 by krwq