dotnet/iot #779: Error 13 initializing the Gpio driver in IoT Edge

cherici-simone, 2019-10-09:
I'm trying to "simply" blink a led in a Raspberry Pi 3 with .net core 3 

**Steps to reproduce**
- follow this https://docs.microsoft.com/en-us/azure/iot-edge/how-to-visual-studio-develop-module to setup the environment
- update runtime and reference in c# module to target .net core 3
- update docker file (see below)
- build/publish/etc.etc.

I also try with different combination (or completely without) of **HostConfig** edge template configuration (see below). The result is always the same.
When I try to run the same docker image outside of IoT edge runtime I got a different error number (**2** instead of **13**).

**program.cs:**
```csharp
static void Main(string[] args) {
            using (var ctrl = new GpioController()) {
                while (true) {
                    ctrl.OpenPin(pin, PinMode.Output);
                    ctrl.Write(pin, PinValue.High);
                    Task.Delay(1000).Wait();
                    ctrl.Write(pin, PinValue.Low);
                    Task.Delay(1000).Wait(); 
                }
            }
        }
```

**docker:**
```docker
FROM mcr.microsoft.com/dotnet/core/runtime:3.0.0-buster-slim-arm32v7 AS base

RUN apt-get update && \
    apt-get install -y --no-install-recommends unzip procps && \
    rm -rf /var/lib/apt/lists/*

RUN useradd -ms /bin/bash moduleuser
USER moduleuser
RUN curl -sSL https://aka.ms/getvsdbgsh | bash /dev/stdin -v latest -l ~/vsdbg

FROM mcr.microsoft.com/dotnet/core/sdk:3.0.100-buster AS build-env
WORKDIR /app

COPY *.csproj ./
RUN dotnet restore

COPY . ./
RUN dotnet publish -c Debug -o out

FROM base
WORKDIR /app
COPY --from=build-env /app/out ./

ENTRYPOINT ["dotnet", "LightHandler.dll"]
```

**csproj:**
```xml
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.0</TargetFramework>
  </PropertyGroup>
  <PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|netcoreapp3.0|AnyCPU'">
    <TreatWarningsAsErrors>True</TreatWarningsAsErrors>
    <TreatSpecificWarningsAsErrors/>
  </PropertyGroup>
  <ItemGroup>
    <ProjectCapability Include="AzureIoTEdgeModule"/>
  </ItemGroup>
  <ItemGroup>
    <PackageReference Include="Microsoft.Azure.Devices.Client" Version="1.*"/>
    <PackageReference Include="Microsoft.Extensions.Configuration" Version="3.0.0"/>
    <PackageReference Include="Microsoft.Extensions.Configuration.Abstractions" Version="3.0.0"/>
    <PackageReference Include="Microsoft.Extensions.Configuration.Binder" Version="3.0.0"/>
    <PackageReference Include="Microsoft.Extensions.Configuration.EnvironmentVariables" Version="3.0.0"/>
    <PackageReference Include="Microsoft.Extensions.Configuration.FileExtensions" Version="3.0.0"/>
    <PackageReference Include="Microsoft.Extensions.Configuration.Json" Version="3.0.0"/>
    <PackageReference Include="System.Runtime.Loader" Version="4.3.0"/>
    <PackageReference Include="System.Device.Gpio" Version="1.0.0"/>
  </ItemGroup>
</Project>
```
**edge template:**
```json
{
  "$schema-template": "1.0.1",
  "modulesContent": {
    "$edgeAgent": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "runtime": {
          "type": "docker",
          "settings": {
            "minDockerVersion": "v1.25",
            "loggingOptions": "",
            "registryCredentials": {}
          }
        },
        "systemModules": {
          "edgeAgent": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-agent:1.0",
              "createOptions": {}
            }
          },
          "edgeHub": {
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-hub:1.0",
              "createOptions": {
                "HostConfig": {
                  "PortBindings": {
                    "5671/tcp": [
                      {
                        "HostPort": "5671"
                      }
                    ],
                    "8883/tcp": [
                      {
                        "HostPort": "8883"
                      }
                    ],
                    "443/tcp": [
                      {
                        "HostPort": "443"
                      }
                    ]
                  }
                }
              }
            }
          }
        },
        "modules": {
          "LightHandler": {
            "version": "1.0.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "${MODULEDIR<../LightHandler>.debug}",
              "createOptions": {
                "HostConfig": {
                  "Privileged": true,
                  "Binds": [ "/dev/gpiomem:/dev/gpiomem" ],
                  "Devices": [
                    {
                      "PathOnHost": "/dev/i2c-1",
                      "PathInContainer": "/dev/i2c-1",
                      "CgroupPermissions": "rwm"
                    },
                    {
                      "PathOnHost": "/dev/gpiomem",
                      "PathInContainer": "/dev/gpiomem",
                      "CgroupPermissions": "rwm"
                    }
                  ],
                  "Mounts": [
                    {
                      "Type": "bind",
                      "Source": "/lib/modules/",
                      "Target": "/lib/modules/"
                    }
                  ]
                }
              }
            }
          }
        }
      }
    },
    "$edgeHub": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "routes": {
          "LightHandlerToIoTHub": "FROM /messages/modules/LightHandler/outputs/* INTO $upstream"
        },
        "storeAndForwardConfiguration": {
          "timeToLiveSecs": 7200
        }
      }
    }
  }
}
```

**Expected behavior**

led blink

**Actual behavior**

```
Unhandled exception. System.IO.IOException: Error 13 initializing the Gpio driver.
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at LightHandler.Program.Main(String[] args) in /app/Program.cs:line 31
```

**Versions used**

- Version of `System.Device.Gpio` package: 1.0.0.0
- Version of `Iot.Device.Bindings` package: not used

ZhangGaoxing, 2019-10-09:
Code 2: Please refer to #764
Code 13: This means "Permission denied". Please run the program with sudo.

cherici-simone, 2019-10-09:
Hello and thanks for tour respinse buy I've already ready and try likes suggested in #764 without success, also with sudo. 
Abiti iot Edge runtime I think It already run with high privilege

RobotTomC, 2019-10-10:
Yes, I have exactly the same error code 👍
Unhandled Exception: System.IO.IOException: Error 13 initializing the Gpio driver.
The only thing I have to add is the the same code worked with dotNet 2.1 but when run as a IoTEdge Module then I get this error.

cherici-simone, 2019-10-11:
An update: running this "chmod 777 /Dev/gpiomem" partially solve the problem...that is I do not have 13 or 2 error anymore. Of course this is not a solution...just a suggestion

RobotTomC, 2019-10-12:
Thanks cherici!
Changing the the permission to "gpiomem" fixed my problem.

joperezr, 2019-10-15:
Closing this one for now since seems like the issue was not running as sudo which we require by design. Feel free to re-open if that is not the case.

closed 2019-10-15 by joperezr