dotnet/iot #784: VS2019 .sln

MarkCiliaVincenti, 2019-10-09:
How possible would it be to have a VS2019 .sln including Iot.Device.Bindings and all the samples so folks using VS2019 on Windows don't need to go through hurdles to contribute to this project? It'd be nice if we could just fork, branch, clone and open in VS, do changes, right click and 'Commit'.

Frankenslag, 2019-10-09:
@MarkCiliaVincenti 
What problems are you facing. I open the folder. Right click on the root project and build the forked folder. It’s not as easy as using a full solution but seems ok.

MarkCiliaVincenti, 2019-10-10:
Well let's see. They say a picture is worth a thousand words.

First I fork dotnet/iot and create my own branch WeatherHelper.
[image]
And click clone & download and "open in Visual Studio"

[image]
Then I click Clone.

Then I have to switch to team explorer
select branches
expand remotes/origin and double click on my branch WeatherHelper
Back to solution explorer
Then from the startup item I choose "Iot.Device.Bindings.csproj"
Then what? If I open files, I won't get syntax errors and I can't F12.

[image]

Frankenslag, 2019-10-10:
@MarkCiliaVincenti 
OK. I do it slightly differently. I don't really know if it is correct as I found conflicting discussions on the web.

* Fork the dotnet/iot into your github area i.e. Frankenslag/iot
* From VS2019 then Clone or Checkout code from Frankenslag/iot to a local folder
* Go into settings for the local repositary and add a remote called upstream with the url https://github.com/dotnet/iot.git. This is useful if you want to keep your local master up to date
* Now from within VS create a new local branch for your current work. Ensure that this new branch is now the current one.
* From within the source explorer right click on the root folder and open a developer command prompt.
* type build.cmd This will make sure all the tooling is in place and up to date within your new branch. This may take some minutes. I see a bunch of errors during the build but I think the important thing is that the tooling has been installed
* right click on the root build.proj and select rebuild to clean and build the projects.

Now when I go into the source code I get syntax highlighting and pressing F12 takes me to a definition etc.

You can sync your new local branch back up to github and then create a PR from there if required.

Other things that I have done is used vsdbg on 32bit versions of raspbian then this allowed me to remote debug on a Raspberry Pi which has been really useful. I have not seen this well documented at all as most of the examples seem to be around visual studio code.

I am really interested in understand how all this 'should' be done however.

MarkCiliaVincenti, 2019-10-10:
I'll give it a go. I feel reassured now that I know that it isn't just me who's struggling with the 'best practice' :)

Clearly, all this doesn't seem intuitive at all, and maybe we should have a tutorial up to show how it should be done, alas we end up discouraging potential contributors.

joperezr, 2019-10-10:
A couple of things to keep in mind as I read this:

- Our repo uses the same build infrastructure than the rest of the dotnet/ repositories (like corefx, coreclr, sdk, etc.), which enables us to build using the same compiler as other repos, get automatic dependencies from other repos, use the same versioning infrastructure than other repos, use the same signing infrastructure, and use the same testing infrastructure, among other things. Because of this, we need to download and restore all of this infrastructure in order for people to be able to build the repo for the first time, which is why we require you running build.cmd/build.sh from the root after cloning before you can open projects in VS.
- We only provided solution for projects that would probably make sense to have open together at the same time. For example, System.Device.Gpio.csproj and System.Device.Gpio.Tests.csproj, which is why we have a System.Device.Gpio.sln. I think you totally have a good point in that we should probably also have .slns for every binding, which would include at least the device binding csproj and its sample csproj. That said, I don't think it is a good idea to have one uber-sln that had all the bindings since I doubt that there will be a lot of times where people will want to work on more than one binding, plus this is a problem that simply won't scale as we keep on getting more device bindings.

krwq, 2019-10-10:
@joperezr I think we should have just one sln because it doesn't hurt since our projects are relatively small. Also CLI provides utils for generating SLN so this can actually be scaled :smile:

I think the main problem here is that we require running `build.cmd`/`build.sh` before using project which is kinda weird comparing to pretty much any other repo everywhere

joperezr, 2019-10-10:
> I think the main problem here is that we require running build.cmd/build.sh before using project which is kinda weird comparing to pretty much any other repo everywhere

I understand but there is no good way of going around this if we want to use the shared infrastructure between other dotnet repos, which we have to.

> I think we should have just one sln because it doesn't hurt since our projects are relatively small.

That is not true for every project, we have some bindings that are quite large, and with time, we will end up having hundreds of proyects in the one solution, and I stand by my claim that I really doubt that will be useful for most people. Very few cases you need to open more than one or two projects when working in a binding. One solution per binding would be the solution IMO.

krwq, 2019-10-11:
@joperezr can we at least add some target which will run build.cmd if it hasn't been run?

joperezr, 2019-10-11:
> can we at least add some target which will run build.cmd if it hasn't been run?

Not really. To run a target you need two things, msbuild (or dotnet CLI) and to be able to load the project. dotnet is restored as part of running build.cmd and in order to be able to load most of our projects you need to have restored the SDKs first.

MarkCiliaVincenti, 2019-10-12:
> @MarkCiliaVincenti
> OK. I do it slightly differently. I don't really know if it is correct as I found conflicting discussions on the web.
>
> * Fork the dotnet/iot into your github area i.e. [...]

I tried all this but it still didn't work. I tried updating a sample but I don't get syntax errors. Then I tried to run some unit tests and I couldn't either.

HakanL, 2020-03-25:
For reference I created this SLN file and put it in the `/src` folder: https://gist.github.com/HakanL/76a5f5c72a34865bbac19c35ec540143
Note that I only added the device and sample that I was working on (Mcp9808), but perhaps this can be used as a template.

JTrotta, 2020-04-01:
> @MarkCiliaVincenti
> OK. I do it slightly differently. I don't really know if it is correct as I found conflicting discussions on the web.
>
> * Fork the dotnet/iot into your github area i.e. [...]

I did all suggested steps.
But when I rebuild build.proj, I got these errors:
[image]

joperezr, 2020-04-01:
@JTrotta Have you first build the repo from the command line? After forking, first thing you need to do before opening any project in Visual Studio is to build the whole repo from the command line, here are the instructions on how to do it: https://github.com/dotnet/iot/blob/a8a2342d9a9660c061a7037d7eb42f9f548f54e7/Documentation/CONTRIBUTING.md#building-from-the-command-line but basically it is simply to call build.cmd (or build.sh on non-Windows) from the root of the repository. After that, you should be able to open individual projects from VisualStudio. If you still have any issues with this please feel free to ping me and I'll gladly help.

pgrawehr, 2020-04-02:
@Jtrotta You might need to go to https://dotnet.microsoft.com/download/dotnet-core?utm_source=getdotnetcorecli&utm_medium=referral and download the matching .NET core SDK (V3.1.1 IIRC). You need to have this installed for VS to work, even if the build script installs a local version into the repository source folder.

JTrotta, 2020-04-02:
@pgrawehr You got it. I supposed that the local version of the SDK would have been enough. Installing the SDK into the system (and then VS), it works.
Thank you!

joperezr, 2020-04-02:
@pgrawehr great catch! we could probably add this info into the contributing doc

Ellerbach, 2020-12-22:
So closing this issue as it seems to be fixed. Feel free to reopen if not.

closed 2020-12-22 by Ellerbach