dotnet/iot #791: I²C in summaries

MarkCiliaVincenti, 2019-10-12:
This is not really a bug, but I've noticed that there's I2C written everywhere in summaries rather than I²C. I started going through them until I noticed there were almost 10,000 lines of code containing "I2C" with match case.

shaggygi, 2019-10-12:
I thought about that when updating the docs, but thought it would be more consistent with others (like SPI) and if that format would cause errors while displaying in different doc formats, IDEs, etc..

I like either approach, but I2C is used lots throughout web, datasheets, etc.

MarkCiliaVincenti, 2019-10-14:
> I thought about that when updating the docs, but thought it would be more consistent with others (like SPI) and if that format would cause errors while displaying in different doc formats, IDEs, etc..
> 
> I like either approach, but I2C is used lots throughout web, datasheets, etc.

To be honest, I'm not a fan of reducing the developer experience just because some devs want to use editors that are so archaic they don't even have Unicode support. If you're developing for a modern programming language like .NET Core, logic dictates that you don't use a code editor from the 1980s. Similarly I was asked to change the degree symbol (°) to `\uXXXX` notation and I complied despite having similar thoughts on it, though to be fair I feel stronger against changing `°C` to `\u00B0C` than I am about having `I2C` instead of `I²C`, because the latter is clearer while `\u00B0C` may be confusing to developers. Basically, I don't think we should make lives difficult to developers who are using a standard that's been around from the early 90s.

krwq, 2019-10-14:
@MarkCiliaVincenti I personally prefer `°` for celsius and for I2C I prefer using just `2` since it's much easier to type and doesn't impede reading. The reason I asked to change is because that's what our coding guidelines are even if I personally disagree with them :(

To be complete fair though I've already seen cases where during some larger changes (moving code from one source control system to another) some of the unicode characters got replaced with question marks and no one noticed that for another several years. Then we noticed it moving to another source base and were curious how could have tests passed that and it turned out the problem was because both input and output replaced special characters into question marks so both actual and expected value had question marks and everything seemed to be passing. While this fortunately hasn't caused any regression we were without proper code coverage for several years. Based on that experience I find this guideline reasonable (and good for test code) even though not preferred.

joperezr, 2019-10-14:
According to some sources, I2c is also correct: "Alternatively I²C is spelled I2C (pronounced I-two-C) or IIC (pronounced I-I-C)." and I2C is also widely found in other libraries that add support for it.

MarkCiliaVincenti, 2019-10-15:
> According to some sources, I2c is also correct: "Alternatively I²C is spelled I2C (pronounced I-two-C) or IIC (pronounced I-I-C)." and I2C is also widely found in other libraries that add support for it.

Definitely, I'm reading I2C quite often, but the correct term is I²C. The former is generally used more because it's easier to write, and less because of Unicode support. Just my $0.02 :-)

joperezr, 2019-10-15:
I understand your point and it is valid. That said, we have already shipped 1.0 using I2c on API names, and I would really love it if all of our docs match the API name so I would still be hesitant to make any changes in code or comments regarding this.

Ellerbach, 2020-03-18:
I understand as well but even NXP is using I2C and not I²C: https://www.nxp.com/docs/en/user-guide/UM10204.pdf
They are those who introduced formally the bus and the name. So I would keep things are they are and won't change the API.

joperezr, 2020-03-18:
I'll go ahead and close this given the above discussion here. Feel free to reopen if you want to continue the discussion.

closed 2020-03-18 by joperezr