dotnet/iot #795: Fast and easy deploy on build to IoT device

maloo, 2019-10-13:
- What is the current best practices for code-build-deploy-run/debug cycle in IoT on an external device? I could not find any relevant information in the repo docs.
- What is the recommended way to build for linux-arm in Visual Studio (e.g. Raspberry Pi)? RuntimeIdentifier manually added to app project? If so, why is it not added by default to projects, or is another use case more common than Raspberry Pi preventing this?
- What is the debugging experience with e.g. a Raspberry Pi? I could not find anything about how to start a debugging session between VS and my app on the Pi in the repo. Just a link to a blog by @shanselman. Maybe these instructions could be formalized and put into the repo docs? Maybe a dotnet tool to set it up on the pi/host by installing the remote debugger support and creating the host launch settings?

I have used ssh-deploy or pure scp in my app projects post build step for fast cycle build-deploy-run. But both of these copy all files at each build, and that can get slow quickly, especially for self-contained apps. I was looking around for alternatives and found ssh.net, and thought maybe we could build a dotnet tool that transfer the changed files to the device post build like ssh-deploy, but only those with updated md5 sum or similar? Or do you have a better solution?

shaggygi, 2019-10-13:
See https://github.com/dotnet/iot/blob/master/Documentation/How-to-Deploy-an-IoT-App.md

maloo, 2019-10-13:
To me deployment is what you do when you are ready to release. Not something you do after each build during development cycle. What I really want is the "new project->speify ip,user,pwd->hit f5" experience. VS C++ has much better support for this type of use case (last time I checked - 2018), and C# should be the "easy" toolchain ;)

joperezr, 2019-10-24:
We should be a bit more clear on what is it that we want to add instructions for here. What you are building for most of the time when you use this library is essentially a .NET Core Console App, so it allows a lot of different deployment mechanisms(self-contained tar ball, shared framework application, single-file executable, docker container with the application, etc.). Additionally, you could also not just develop a console app, but might be developing a library instead that uses GPIO internally, or you could also be trying to run a web service on the raspberry Pi.

In summary, the deployment mechanisms this library has are the exact same deployment mechanisms that .NET Core has, which are many. The only thing different about this repo, is that during development, you might be developing on a faster dev machine, but running on the raspberry pi, which is why we includded the docs that @shaggygi pointed out in his comment.

maloo, 2019-10-24:
Agreed, it is the development cycle that should be improved. And deployment is the same as for any other app.
My main point is that just a simple add on like ssh copy task during build and a better debug setup will provide a far better experience than we have now. Maybe parts of the c++ tooling could be reused.

krwq, 2023-01-26:
[Triage] We should opportunistically add docs, please create PR if you have any suggestions.

closed 2023-01-26 by krwq