dotnet/iot #798: Failing unit tests might not print the output

krwq, 2019-10-15:
FYI in case someone hits this.

With https://github.com/dotnet/iot/pull/789 build.cmd/build.sh will run unit tests but unfortunately due to bug in VSTest the output most of the time will not get printed (depends which worker picks up the job).

This is causing that if any of the unit test fails the output might not get printed and it's hard to diagnose what went wrong. Most of the time just passing `-v d` will fix the issue. Using `dotnet test` always prints the output correctly.

VSTest issue: https://github.com/microsoft/vstest/issues/2224

MarkCiliaVincenti, 2019-10-15:
Maybe a stupid question from my end, but can't the build.cmd/build.sh run the `dotnet test` at the end?

joperezr, 2019-10-15:
> but can't the build.cmd/build.sh run the dotnet test at the end?

Very valid question. Ideally we want the build of the whole product to be part of the same `dotnet build` evaluation. This way, allprojects are only loaded once and get evaluated once, and targets that only need to run once per full build will do so only once, since if you call `dotnet test` in the end what will happen is that a bunch of stuff that was already built, will get rebuilt. That isn't super terrible in our specific repo, but given this is a bug in the tester, instead of working it around in our end, it is better to track the resolution of the root cause in order to make our tools better for everyone.