dotnet/iot #805: Running build.sh on the Raspberry Pi fails with "wrong ELF class: ELFCLASS64"

pgrawehr, 2019-10-16:
**Describe the bug**
Running "./build.sh" from the working copy root on the Pi4 (Raspbian 32 Bit) fails with a bunch of errors of the form 
```
/home/pi/.nuget/packages/microsoft.net.compilers.toolset/3.1.0-beta1-19127-06/tasks/netcoreapp2.1/Microsoft.CSharp.Core.targets(58,5): error : An error occurred while loading required library libhostpolicy.so from [/home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App/2.1.11] [/home/pi/projects/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
/home/pi/.nuget/packages/microsoft.net.compilers.toolset/3.1.0-beta1-19127-06/tasks/netcoreapp2.1/Microsoft.CSharp.Core.targets(58,5): error : Failed to load ��(, error: /home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App/2.1.11/libhostpolicy.so: wrong ELF class: ELFCLASS64 [/home/pi/projects/iot/src/devices/Units/Units.csproj]
```
The linux distribution installed on the Pi is 32 bit ARM, so this seems to use a wrong library. 
Note: Running `cd src/System.Device.Gpio && dotnet build System.Device.Gpio.sln` works as expected. 

**Steps to reproduce**
See above. Type
```
./build.sh
```
in the project root folder. 

**Expected behavior**

Build finishes without errors. 

**Actual behavior**

Build fails with exit code 1

**Versions used**

Built against these versions (but doesn't really seem to matter):
```
.NET Core SDKs installed:
  3.0.100 [/home/pi/projects/iot/.dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/home/pi/projects/iot/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.11 [/home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [/home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App]
```

pgrawehr, 2019-10-16:
Here's the full build log:

[buildPi4.log](https://github.com/dotnet/iot/files/3736052/buildPi4.log)

pgrawehr, 2019-10-16:
Note: The unprintable characters are also unprintable here.

krwq, 2019-10-16:
@pgrawehr can you print full output of `<your IOT path>/.dotnet/dotnet --info`? I'm specifically curious about Runtime Environment section.

pgrawehr, 2019-10-17:
Here's the full printout: 
```
pi@raspberrypi:~/projects/iot $ .dotnet/dotnet --info
.NET Core SDK (gemäß "global.json"):
 Version:   3.0.100
 Commit:    04339c3a26

Laufzeitumgebung:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/projects/iot/.dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  3.0.100 [/home/pi/projects/iot/.dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/home/pi/projects/iot/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.11 [/home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [/home/pi/projects/iot/.dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
pi@raspberrypi:~/projects/iot $ 

```

The runtime section doesn't mention ARM32 or ARM64, so maybe that's part of the problem?

Ellerbach, 2020-12-22:
@pgrawehr is this fix now?

pgrawehr, 2020-12-22:
I need to test again. Last time I tried it failed with another error, IIRC.

pgrawehr, 2020-12-26:
Just tested again: The build works fine, it just has a small glitch: Running (some?) test cases fails, because the dotnet SDK 2.1 is not automatically installed into the build folder. After

```
pi@raspberrypi:~/Downloads $ tar xzf dotnet-sdk-2.1.811-linux-arm.tar.gz -C ~/projects/iot4/.dotnet
```
everything was green. 

So I'm closing this as fixed with a minor issue left.

closed 2020-12-26 by pgrawehr