dotnet/iot #807: Add dogfooding instructions

krwq, 2019-10-17:
Add instructions how to use latest package from master (including screenshots for VS instructions).

trevorjobling, 2019-10-30:
This would be so useful right now. If get to the bottom of this I'll contribute dogfooding instructions based on my experience!

I've cloned the `iot` repo. It builds successfully using `.\build.cmd`. 

I'm referencing the Gpio project in my dotnetcore 3.0 project using
```
<ProjectReference Include="..\iot\src\System.Device.Gpio\System.Device.Gpio.csproj" />
```
When I `dotnet build` by project I can't get past this error:

`C:\Program Files\dotnet\sdk\3.0.100\Roslyn\Microsoft.CSharp.Core.targets(59,5): error MSB4062: The "Microsoft.CodeAnalysis.BuildTasks.Csc" task could not be loaded from the assembly C:\Program Files\dotnet\sdk\3.0.100\Roslyn\Microsoft.Build.Tasks.CodeAnalysis.dll. Assembly with same name is already loaded Confirm that the <UsingTask> declaration is correct, that the assembly and all its dependencies are available, and that the task contains a public class that implements Microsoft.Build.Framework.ITask. [C:\scc\ShamboShed\ShamboShed\ShamboShed.csproj]`

Is this familiar to anyone? My google-fu is failing me and I'm stuck.

joperezr, 2019-10-30:
Is there a specific reason why you would want to consume the library via a project reference? If you are not making private changes on System.Device.Gpio I would recommend either just having your dotnetcore 3.0 project reference the nightly build of System.Device.Gpio package, or to run `build.cmd -package` when you are building the repo in order to generate the packages. For the first option (referencing our nightly build), here are the instructions:

1. Add our nightly build source into your project so that you can reference our nightly builds
You can accomplish this by running `dotnet new nugetconfig` on the directory where your dotnetcore 3.0 project was created, and manually editing the generated NuGet.Config in order to add a new source: `<add key="dotnet-iot" value="https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json" />`
2. Add a reference to the latest version of the package into your .csproj. You can either manually specify a version, or you could alternatively simply always get the latest by adding a reference like:
```xml
<ItemGroup>
  <PackageReference Include="System.Device.Gpio" Version="1.1.*" />
</ItemGroup>
```
3. Run `dotnet restore`
4. That should be it, you should now be able to use the latest changes inside your dotnetcore3.0 project.

Let us know if that doesn't work for you.

trevorjobling, 2019-10-31:
That worked a great - thanks! Much better than what I was trying.

BTW this didn't work for me
`<PackageReference Include="System.Device.Gpio" Version="1.1.*" />`

Changing to this avoids errors regarding _unable to find package with version (>= 1.1.0)_
`<PackageReference Include="Iot.Device.Bindings" Version="1.1.0-p*" />`

joperezr, 2019-10-31:
Oh you are right! I got the version wrong, sorry about that, I will fix my comment so that people in the future will have the right instructions form this issue before we move it to actual documentation

krwq, 2019-11-06:
note that it is recommended to use specific version of the package: in case we make any breaking changes it won't break your app

pgrawehr, 2022-12-10:
Closing this as no longer an issue, as everything works fine when using nuget. Generic questions about the use of different versions of packages are out of scope here.

closed 2022-12-10 by pgrawehr