dotnet/iot #809: API Suggestion for Low-Level GPIO Drivers

pgrawehr, 2019-10-19:
Since the issue addressed in #788 seems to require wider discussion, let's talk about how it should look like to the user in the end (and how that concept should be implemented).

# Overview of the Problems
The primary problem is that the Raspberry Pi4 requires a few things (in particular: Setting the pull up/pull down mode of input pins) to be done differently than on the Pi3. (Not sure about even older models, and whether we support them.)

This leads to two main topics: 
1. How do we organize the two drivers? Which classes/methods should be exposed to the user?
2. How do we detect the correct driver to use? Should we allow the user to manually select a driver? Should there be a way of finding out whether a driver matches a given platform?

# Problem 1: Driver class structure
The current proposal is
```
public abstract class GpioDriver;
    public abstract class GenericRaspberryPiDriver : GpioDriver;*
        public class RaspberryPi3Driver: GenericRaspberryPiDriver;
        public class RaspberryPi4Driver: GenericRaspberryPiDriver;
   public class SysFsDriver: GpioDriver;
   ....

* Note: Additional intermediate class on Windows: Windows10Driver
```
This would expose a new abstract GenericRaspberryPiDriver, as well as the new RaspberryPi4Driver. 

# Problem 2: Detect the correct driver to use / select the desired driver
## Situation
Currently, GpioController has an internal static method `GetBestDriverForBoard()` which returns (after some quite ugly magic, see also #766) a reference to the best matching `GpioDriver` for the current hardware/operating system. `GetBestDriverForBoard()` is called when `GpioController` is created without explicitly specifiying a driver. The user can, however, create instances of any of the Driver classes and use them as argument to the GpioController constructor. 

## Issues with the current solution
The downside of this approach is that the user doesn't know what he actually gets when he uses the BestDriver approach. When he wants to use the "second best" driver (for whatever reason, maybe for testing), he doesn't know which to use. 
If a selected driver doesn't work (be it automatically selected or manually), there's no defined way of knowing, unless probing and hoping not to get an exception (or worse, an access violation). 

## Suggestion
My suggestion would be to add a `CheckDriverIsCompatible()` method to each driver, indicating whether this driver is supported on this hardware (and with the currently installed software components). Then, we only need to specify somewhere a preference order in which `GetBestDriverForBoard()` probes them. In the concrete example, the Pi3Driver would only report true on the Pi3, the Pi4 driver on the Pi4. This could also allow a cleaner solution to #766. 

Then, we could expose the driver list (in preference order) using a static method, but hide all the drivers themselves. Since except for the constructor and the destructor, nothing is public of these classes, the user cannot really do anything with them, anyway. And externally deriving (to get access to the protected methods) is also not really useful, since any added functionality would need access to the low-level memory pointer or handle (which absolutelly should stay internal).

Frankenslag, 2019-10-19:
@pgrawehr 

Regarding the hierarchy. I am not sure that we need a proposal for that. I feel it is down to the magnitude of differences between models and hardware. Personally I feel that if the difference between the pi3 and pi4 comes down to the pullup registers then it should be part of a single driver. So I would go for whatever hierarchy was appropriate.

I guess that the SYSFS and Unix, Windows 10 and Libgpiod driver would also return true in certain circumstances. What prioritization would mechanism would be used ?

pgrawehr, 2019-10-19:
If we don't go with two drivers for Pi3 and Pi4, then at least the one single driver should change its name. As pointed out in the PR, the additional registers required on the Pi4 don't exist on the Pi3, so using the same driver may lead to mapping addresses that don't exist, which is not optimal. 

I'm not sure about the prioritization. It'll probably be hardcoded in a "most specific" to "most generic" order. And yes, the SysFs and Libgpiod driver do work as well, although they may have limited features (i.e. setting the pulls is explicitly not supported by libgpiod) or require root rights.

ZhangGaoxing, 2019-10-20:
1. I think we should abstract the generic control method for a series of SoCs, instead of just naming it `XxxPiDriver`, like I did in #780 . Because it's not just a board that uses this SoC. When adding new drivers, we only need to make small changes, such as pin mapping, address offset, etc.
2. For the second question, I have a direct idea. We don't need regular expressions to get CPU information. We just need to support the official system. It is not necessary and possible to support all systems. The official system can directly read the name of the current board, and return the best driver according to the name. For unofficial systems, users can also manually select the appropriate driver to achieve the function.

pgrawehr, 2019-10-20:
The conceptual question in this issue is not how to detect a specific CPU/board type or revision, but how to organize the detection and how the user should select the correct driver. If the classes are user-visible, they need to have names he can recognize. And the drivers will have board specific features.

pgrawehr, 2019-10-23:
@joperezr : I'm fine if you don't have time to discuss this right now, but can you please add this to vNext? I think we should definitelly have a fix for this problem for the next release.

Frankenslag, 2019-10-23:
@pgrawehr 
I think we still have a bug to address though and that should probably not wait for vNext. If you want I could come up with something by modifying the existing raspberry pi driver. Please let me know.

pgrawehr, 2019-10-23:
I could do that myself as well, I just need a hint on the way forward. If it is ok to patch the existing driver (so we don't change the API) I'm fine with that for now. Is there a documentation / ticket describing what is expected to be in which patch/minor/major/feature release? I'm struggling a bit to understand the expected release cycle and the task priorities.

Frankenslag, 2019-10-23:
@pgrawehr 
I think you patch so the api doesn't change. The changes to how the Pi4 works haven't been broadcast widely but they could impact quite a few people I believe and so a bug fix patch would be really good. Hopefully we can get it rolled into the code base in time for 3.1

I haven't seen anything describing what would be a minor/major/feature etc. There is a roadmap [Here](https://github.com/dotnet/iot/blob/ee31c42e840786aa337e331c7c7e2cbd57dd8e8b/Documentation/roadmap.md)

pgrawehr, 2019-10-23:
Ok, I'll come up with a patch fix, then. 
Also, we should be clear about version numbers, I guess. The currently released version caries 1.0.0.

krwq, 2019-11-26:
I agree that current class hierarchy is a bit problematic and not sure at this point how this will scale out in the future. As for `CheckDriverIsCompatible` - I'm not sure this is a good idea - I can imagine a driver where giving the answer is not possible and throwing exception or returning in such method would be inapropriate/incorrect.

Long term I'd like to see board not differ much to any other multi-device we currently support, i.e. SenseHat or Pca9685 (not exactly multi device but exposes multiple PWMs) but regardless of the design I think there will be a need for some common abstraction which is what this issue is trying to solve. I don't think we currently have enough data to make the right decision. I suggest let's keep working on support for new boards/drivers and the abstraction should over time become obvious. I do not want to end up with really inconvenient and annoying to maintain API.

What I think will happen is that boards will eventually tend to the shape similar to:

```csharp
class RaspberryPi125Board
{
  public I2cDevice CreateI2cDevice(...); // similar to I2cDevice.Create but more explicit
  public PwmChannel CreatePwmChannel(...);
  public GpioController CreateGpioController(...);
  public Temperature GetCPUTemperature();
  // ...
}
```

but I do not want to get too much ahead here since whatever design we come up with right now will probably be broken by some unusual board in the future - IMO waiting here is better idea - note especially with System.Device.Gpio once we add API we will likely be stuck with it forever so better to be a bit conservative about what we add.

pgrawehr, 2019-11-27:
@krwq : I think that approach is probably a good idea, as it would also allow solutions for problems such as #857, where creating a device (i.e. a PWM channel) requires configuration of some other device (gpio pins in this case). It could also support error checking, such that one cannot open the I2C bus and the respective GPIO pins at the same time. 

Also, this allows creating a board object and then working with board-relative device addresses. This could be handy i.e. for #262, where we would only need to define the connection to the arduino once and then create it's devices. Imagine the confusion we have with the current interface if you where working on a Pi with an attached arduino and you want to open the arduino's PWM device...

I agree that a clean solution should be well discussed, but waiting for all possible boards is not really possible, I think.