dotnet/iot #813: How to use Hardware PWM?

pgrawehr, 2019-10-20:
I cannot find any documentation on how to use hardware PWM on the Raspberry Pi, and I cannot get it to run. 

Enabling PWM requires setting the Pin Mode for (BCM) Pins 12 or 13 to ALT0 mode, for which the library does not offer a possibility, it seems. (Note: Unlike most other alternate functions, PWM mode does not require a special driver and therefore also not a reboot to activate). 

I'm also not sure whether the currently implemented method of using /sys/class/pwm is the right one to access the PWM device. This directory is always empty, even if I enable PWM using `gpio -g mode 13 pwm`. 

Note: The following works to test the PWM Channel (attach an LED to BCM Pin 13):
```
gpio -g mode 13 pwm
gpio -g pwm 13 <value between 0 and 1024>
```

krwq, 2019-10-21:
I'm closing this as a dup of https://github.com/dotnet/iot/issues/802

Since the other issue doesn't have the answer.

To enable PWM use:

I recommend to read /boot/overlays/README since it describes pretty much how to enable anything on the PI.

pgrawehr, 2019-10-21:
@krwq: Ok, I'll try this approach, but I'm not happy with it. Some investigation showed that it's possible to drive the PWM mode from memory mapped registers. This, as said, has the advantage that it does not require an overlay module to be loaded and pin assignments can change without reboot. So we should have a driver for that. 
In either case, we need a way of setting the GPIO pins to alternate functions (and back). Even if the overlay mode was properly loaded, some (other) application might have set pin modes in such a way that the requested function (i.e. I2C) is not currently available.

joperezr, 2019-10-21:
@pgrawehr We could do that but the problem is that this would only work for one board. You are correct in the fact that using registers and setting alternate functions on pins you are able to control PWM as well, but we thought that if we could come up with a non-board-specific implementation of PWM would be ideal. Is there a specific reason other than requiring another overlay module loaded for which you prefer the register way?

krwq, 2019-10-21:
@joperezr we could potentially expose it directly from RaspberryPiDriver (i.e. CreatePwmChannel)

joperezr, 2019-10-21:
If we find a good use-case for it, for which the generic driver is not sufficient then yes this is one option. Before adding extra code we would have to maintain and wouldn't be portable for the rest of the boards, I would like to make sure that there is a good reason why we it.

pgrawehr, 2019-10-21:
One reason one may not want the PWM overlay enabled is that this disturbs the audio output (at least according to the docs, I haven't tried yet). I need to do more testing to see the implications of this setup. 

@krwq : That's exactly what I would've suggested as well, however I think this would not even be neccessary, if we just offered a "RaspberryPiPwmDevice". 

Nevertheless, I think we need to extend the PinMode enum to set Alt0-Alt5 modes (for boards that support them). This wouldn't even be a breaking change for existing drivers, since they already throw on anything but Input or Output.

joperezr, 2019-10-21:
> One reason one may not want the PWM overlay enabled is that this disturbs the audio output

I believe this you will hit no matter what. Last time I checked this, the problem was that the resources used by hardware PWM are shared with the audio, so you can't use them at the same time. In fact, IIRC when I originally was writing the RaspberryPi driver I saw a comment on the PWM registers saying that they would conflict with Audio.

pgrawehr, 2019-10-21:
@joperezr : As said, I haven't tried what they actually mean by "conflict". Unfortunatelly, I still can't get PWM to work with this library. If I enable the overlay, I get the (virtual) directories /sys/class/pwm and /sys/class/pwm/pwmchip0 created (and the pin mode defaults to Alt0), but there's no pwm{channel} subdirectory there (as suggested by UnixPwmChannel.Linux.cs, Line 40). I don't get an exception, but just nothing happens. 

One other thing I found, though: It looks like this method is requiring root rights, while the register method would most likely not. That would IMHO be a strong reason to add support for that.

joperezr, 2019-10-21:
The register method would require root permissions as well. Right now RaspberryPiDriver doesn't require root because we read /dev/gpiomem/ memmory mapped file, which only gives you access to the gpio registers. If you want to access to the rest of the registers (which contain the PWM registers as well) then you will need to read /dev/mem/ file which requires root. I want to clear out that is not that I'm opposed to doing this, I just actually had a branch where I had already added all of the code for hardware use of PWM using registers which is kind of why I know all these things, and didn't see a single feature difference over the generic driver(other than requiring to enable the overlay module) which is why originally we decided not to have it. I'm open to discussion though in case there actually is an advantage, which is kind of why I wanted to understand your ask a bit more 😄

krwq, 2019-10-21:
@pgrawehr I'm also curious about the non-sudo solution. If you think it's possible to do with gpiomem or some other way I think we should support such driver.

pgrawehr, 2019-10-22:
@joperezr : Can you post the code you already did? I'll try whether I find a way around the sudo requirement. 
[I thought that the test with the gpio tool would prove that it's not required, but I just saw that this has the setuid bit set, so it always executes as root.]

joperezr, 2019-10-22:
@pgrawehr This was some time ago and in a different repo, but here is the commit: https://github.com/joperezr/corefxlab/commit/970f4a57047e53ab22ec972e303307c0afca745d

The part I definitely remember is that gpiomem doesn't have access to the PWM registers, and also that the /dev/mem requires sudo access so I don't think doing this through registers without sudo will be possible.

Frankenslag, 2019-10-22:
> The part I definitely remember is that gpiomem doesn't have access to the PWM registers, and also that the /dev/mem requires sudo access so I don't think doing this through registers without sudo will be possible.

@joperezr 

Yes. When I look at gpiomem then the length of the mapping is BLOCK_SIZE which isn’t big enough to include the PWM registers.

krwq, 2019-10-22:
@Frankenslag might be worth considering contributing to the linux kernel then even if we can't have it now then we might have it in a year or two

pgrawehr, 2019-10-22:
Just did a quick test (it's time for bed here): a bit of `sudo chmod 666 /sys/class/pwm/pwmchip0` (and a few more of the virtual files there) made it work with the existing driver as non-root. And made it work at all for me for the first time. Continuing investigations tommorow. 

(Yes, I know this is a BIG Hack...)

krwq, 2019-10-22:
I guess using udev to auto-set permission is fine workaround for now

joperezr, 2019-10-22:
> (Yes, I know this is a BIG Hack...)

😄 yeah, that would work but unfortunately we can't really do that.

pgrawehr, 2019-10-24:
I don't know how exactly udev would need to be configured, but I guess that's something that could be requested from the raspbian guys. Many other raspberry tools also seem to use that approach.

krwq, 2019-11-07:
@pgrawehr you just need to modify one file so that permissions get autoset whenever device shows up. I can help with that. I have mixed feelings this should be done in Raspbian

pgrawehr, 2019-11-07:
Ok, thanks for the hint. I'll try this and report back.

pgrawehr, 2019-11-16:
Here's the required udev file, ready for copy-pasting: 
https://github.com/raspberrypi/linux/issues/1983.
They even implemented a kernel fix to make this work, but
why the file (or the file change) itself is not part of the 
kernel, I don't know. 
Now testing.

krwq, 2019-11-18:
@pgrawehr btw. the files from this repo are GPL v2 which is not compatible with MIT which this repo is in

Ellerbach, 2020-12-22:
@pgrawehr I added documentation quite some time ago how to setup hardware PWM on Raspberry Pi and also adjust the security to avoid having to run the code using sudo. Check https://github.com/dotnet/iot/blob/master/Documentation/raspi-pwm.md and if that answer the question, feel free to close the issue.

pgrawehr, 2021-01-27:
Closing, as this has been fixed in the latest kernel versions.

closed 2021-01-27 by pgrawehr