dotnet/iot #817: Display SSD1306

alberk8, 2019-10-22:
**Is your feature request related to a problem? Please describe.**

Describe what problem are you trying to solve or what's the expected outcome.
I have tied the sample code on a SSD1306 128x64 screen (0.96") and it works.  But I does not seems to be able to display a bitmap of size 128x64, only the top 32 pixels (height) are shown.  Is there any settings that I need to set to get the full picture shown?.  Thank you. 

If you'd like us to add a new device binding please add a full name and ideally links to specification / pictures of device you'd like to see here.

Please review [list of devices](https://github.com/dotnet/iot/blob/master/src/devices/README.md) to see if there is any similar device (perhaps belonging to the same family) which can be extended to also support your device.

**Describe the ideal solution**

A clear and concise description of what you want to happen.

krwq, 2019-10-22:
cc: @DemoS76

Frankenslag, 2019-10-23:
@alberk8 cc @DemoS76 

I have had a quick look and I would say that the sample is hard wired to deal with 128 x 32 images.

I think that a good place to start would be to change the line in DisplayImage method that sets the number of pages to be copied

From :
            `Int16 pages = 4;`
To:
           ` Int16 pages = 8;`

There may be more to do than this but I cannot find me SSD1306 to verify.

alberk8, 2019-10-30:
I have tried all the suggestions and even got another unit of SSD1306 from another vendor to test.  It still does not show the 128 x 64.  Using python sample code on the same SS1306 does show image in the 128 x 64 resolution.
So the problem is in the C# code.  I am not sure what else I need to adjust to get that working in the right resolution.

Frankenslag, 2019-10-30:
> I have tried all the suggestions and even got another unit of SSD1306 from another vendor to test. It still does not show the 128 x 64. Using python sample code on the same SS1306 does show image in [...]

Frankenslag, 2019-10-30:
Ok I have got one on order... will have a look over the coming days and get back to you

alberk8, 2019-10-30:
Thank you.  I have also on order SSD1327 and a bit worried as I see the comments on the sample code indicating the 96 x 96 but the specs sheets for SS1327 mention it is a 128 x 128.   Hope it is a typo.

Frankenslag, 2019-10-30:
> Thank you.  I have also on order SSD1327 and a bit worried as I see the comments on the sample code indicating the 96 x 96 but the specs sheets for SS1327 mention it is a 128 x 128.   Hope it is a typo.

The way I read is that the SSD1306 can be used to interface to a number of OLED displays of various resolutions. The one in the sample happens to be 128 x 32 but it should work with displays up to 128 x 64 bits. Have you got a link to the working Python.

I believe it's a similar story for the SSD1327 in that the max resolution will depend on which display is coupled with it.  Which one have you ordered exactly ?

alberk8, 2019-10-31:
For the Python code, I use this link https://circuitdigest.com/microcontroller-projects/ssd1306-oled-display-with-raspberry-pi

I have ordered the SSD1327 from Aliexpress site.  https://bit.ly/2BXYDv7

krwq, 2019-11-01:
@alberk8 unfortunately the first link you shared doesn't allow non-commercial use so we cannot use it as a base for our implementation

Frankenslag, 2019-11-03:
OK. I have got an Ssd1306 connected with a 128 x 64 OLED connected. What I see is that the first 32 lines are 'stretched' over the 64 lines. Examining it closer it looks like only every alternate line is used. This sort of makes sense I think. So I believe that what needs to be done is change the initialization to suit the OLED and also change the drawing functions in the sample.

Frankenslag, 2019-11-03:
I have now modified the sample to work with the 128 x 64 OLED as follows 

The initilalization routine was modified. This takes no account of any different electrical characteristics of the OLED. It does work for me.

* `device.SendCommand(new SetMultiplexRatio(0x3F));` Wrap around at the 64th line
* `device.SendCommand(new Ssd1306Cmnds.SetComPinsHardwareConfiguration(true, false));` Configure for an alternate common line arrangement
* `device.SendCommand(new Ssd1306Cmnds.SetPageAddress(Ssd1306Cmnds.PageAddress.Page0, Ssd1306Cmnds.PageAddress.Page7));` Use 8 pages of buffer memory

A different ClearScreen, DisplayClock and DisplayImage method to account for the OLED height and potentially different height of the images.

        private static void ClearScreen(Ssd1306 device)
        {
            device.SendCommand(new Ssd1306Cmnds.SetColumnAddress());
            device.SendCommand(new Ssd1306Cmnds.SetPageAddress(Ssd1306Cmnds.PageAddress.Page0, Ssd1306Cmnds.PageAddress.Page7));

            for (int cnt = 0; cnt < 64; cnt++)
            {
                byte[] data = new byte[16];
                device.SendData(data);
            }
        }
        private static void DisplayClock(Ssd1306 ssd1306)
        {
            Console.WriteLine("Display clock");
            var fontSize = 25;
            var font =  "DejaVu Sans";
            var fontsys = SystemFonts.CreateFont(font, fontSize, FontStyle.Italic);
            var y = 0;

            foreach(var i in Enumerable.Range(0,100))
            {
                using (Image<Rgba32> image  = new Image<Rgba32>(128, 64))
                {
                    image.Mutate(ctx => ctx
                        .Fill(Rgba32.Black)
                        .DrawText(DateTime.Now.ToString("HH:mm:ss"), fontsys, Rgba32.White, new SixLabors.Primitives.PointF(0, y))
                    );

                    using(Image<Gray16> image_t = image.CloneAs<Gray16>())
                    {
                        ssd1306.DisplayImage(image_t);
                    }

                    y++;
                    if(y >= image.Height) y = 0;

                    Thread.Sleep(100);
                }
            }
        }
        public static void DisplayImage(this Ssd1306 s, Image<Gray16> image)
        {
            Int16 width = 128;
            Int16 pages = (Int16)(image.Height / 8);
            List<byte> buffer = new List<byte>();

            for (int page = 0; page < pages; page++)
                {
                    for (int x = 0; x < width; x++)
                    {
                        int bits = 0;
                        for (byte bit = 0; bit < 8; bit++)
                        {
                            bits = bits << 1;
                            bits |= image[x, page * 8 + 7 - bit].PackedValue > 0 ? 1 : 0;
                        }

                        buffer.Add((byte)bits);
                    }
                }

            int chunk_size = 16;
            for(int i = 0; i < buffer.Count; i += chunk_size)
            {
                s.SendData(buffer.Skip(i).Take(chunk_size).ToArray());
            }
        }

I think that the takeaway from this is that the sample if designed for a very specific piece of hardware and deviations from that require changes to the sample to work properly.

Ellerbach, 2020-12-22:
SSD13xx family added: https://github.com/dotnet/iot/tree/master/src/devices/Ssd13xx
so closing this issue. Feel free to reopen if needed.

closed 2020-12-22 by Ellerbach