dotnet/iot #819: SPI TFT Screen for ILITEK

alberk8, 2019-10-23:
**Is your feature request related to a problem? Please describe.**

Is there any plan development for the ILITEK SPI display chipset ILI9341 and ILI9486 ?  Thank you.

krwq, 2019-10-23:
I have ILI9341 code somewhere in my private branches and can add basic support relatively quickly, it just needs some clean up. I'm kinda busy this week and early next week so please remind me if you don't see any PR in the next two weeks

krwq, 2019-10-23:
Just made it public: https://github.com/krwq/iot/tree/ILI9341 is my WIP branch from half a year ago in case you need it now

alberk8, 2019-10-23:
Thank you so much.

alberk8, 2020-01-07:
> I have ILI9341 code somewhere in my private branches and can add basic support relatively quickly, it just needs some clean up. I'm kinda busy this week and early next week so please remind me if you don't see any PR in the next two weeks

Is the PR going to happen soon?  Thanks.

krwq, 2020-01-07:
@alberk8 haven't had time yet but this is really high on my TODO list at this point so might happen soonish

davidjmallar, 2021-05-31:
Is it soonish yet? :)

krwq, 2021-06-01:
Sorry, I don't think I'll find time for this anytime soon... I've moved out since then, the TFT screen is in one of the boxes somewhere which I haven't unpacked yet and I don't really have time to do this unless I find some interesting project which needs the screen... Feel free to pick up my branch and go from there if you feel like it... I'll update this thread if something changes. If you start working on this please write so in the thread so we don't accidentally duplicate.

dotMorten, 2021-08-19:
@krwq I took your branch and updated it a bit to better match how the other displays work. Would you mind if I continued your work and made a PR for it? (I'll fork from yours so you still get some of the credit)

krwq, 2021-08-19:
go ahead, having a binding matters more than credit 😄

dotMorten, 2021-08-19:
Awesome! PR linked above.

krwq, 2023-01-19:
[Triage] ILI9341 is already merged. Closing this issue since we're not tracking binding requests (send PR 😄)

closed 2023-01-19 by krwq