dotnet/iot #821: DCMotor example improvement

wwombatt, 2019-10-23:
Hello,

I am trying to use the DCMotor example (https://github.com/dotnet/iot/tree/master/src/devices/DCMotor/samples) to drive to motors using the H Bridge component as used in the example but i cannot figure out how to connect the H bridge to a raspberry pi to make it work with the pinnumbers used. I used the information from issue 478 (https://github.com/dotnet/iot/issues/478) to try and make sense out of it but it still is not possible. 

in the example it uses this code `DCMotor motor = DCMotor.Create(6, 26, 21)`

so pinnumbers 6 26 and 21 which following the info in issue 478 it uses the logical pin numbers so 

pin bcm6 is physical pin location 31
pin bcm26 is physical pin location 37
pin bcm21 is physical pin location 40

This does not make sense if you look at the frtitzing diagram. Is there any possibility for clarification?

krwq, 2019-10-23:
@wwombatt I think I might have accidentally updated pin numbers when I was doing the last changes and diagram left from the previous version. It should match whatever is on the picture or picture should be updated.

The pins passed in are BCM numbering

wwombatt, 2019-10-24:
@krwq I am trying to figure out how this example diagram would work with two three pin motor set up. This is what i have so far. Based on the pinout info and the diagram i figure i would neet the following pinnumbers: BCM23 BCM24 BCM25 BCM5 and BCM6. But how would you know which pin to use where when trying to create a motor object via ' DCMotor.Create'. Could you maybe tell me what the code should be when you want to create to motor outputs on the H bridge ?