dotnet/iot #826: Does .NET Core support Raspberry Pi 4?

Penergy, 2019-10-27:
I have seen the instructions for .Net Core on RPi 2 and RPi 3 from this link.
https://github.com/dotnet/core/blob/master/samples/RaspberryPiInstructions.md

I want to know whether .Net Core has the plan to support RPi 4. Because we are going to use RPi 4 later.

shaggygi, 2019-10-28:
> Hey guys. Any chance that pi4 would support sending the command to the USB connected printer from NET Core? I am interesting about Zebra printers and ZPL commands in particular

@vlyakh Referencing https://github.com/dotnet/wpf/issues/871

krwq, 2019-11-06:
@vlyakh this is the first request we have about printers. I'd recommend exploring if there is already any .NET library anywhere which already supports it, perhaps: https://github.com/germanger/zpl-printer or https://github.com/rkone/sharpzebra will do exactly what you need.

If that doesn't work for you please create a feature request on https://github.com/dotnet/iot - note that we are greatly relying on community help with features so this might not get any traction for some time.