dotnet/iot #835: Support for Meadow F7

ethos-tim, 2019-11-01:
I would love to be able to use this package with the Meadow F7

https://www.wildernesslabs.co/meadow

https://store.wildernesslabs.co/collections/frontpage/products/meadow-f7

http://beta-developer.wildernesslabs.co/docs/api/Meadow/

https://github.com/WildernessLabs/Meadow.Foundation

It seems fragmented that everyone would have a different library and duplicate all the device support in .NET vs. having one project. 

I have no experience interfacing with hardware but would love to help. Is there a technical reason the board can't be support?

joperezr, 2019-11-11:
cc: @richlander 

Hey @ethos-tim thanks for logging this issue. We do agree that it is a great idea to work together in order to better align to that ecosystem. @richlander can probably provide more details about this.

frogcrush, 2020-11-11:
I also agree that this sort of alignment would be great for the community; being able to target one API and carry it across multiple controllers sounds pretty great, and hopefully there would be less duplicate work when writing support for devices. I can't see a downside to not having to port between three different platforms (nanoFramework, this, and Meadow) as long as the underlying hardware supports it.

krwq, 2023-01-19:
[Triage] We're not actively tracking binding/driver requests. Someone who has the board will need to write a driver and we can review the PR. If there is anything to discuss please re-open