dotnet/iot #836: Close 3.0 Milestone

shaggygi, 2019-11-02:
[3.0 Milestone](https://github.com/dotnet/iot/milestone/1) can now be closed.

pgrawehr, 2019-11-02:
As a side note: The currently released System.Device.Iot nuget package (and also the file inside it) has Version 1.0.0. This should be brought in sync for the next release.

joperezr, 2019-11-04:
You are right it may be closed. That said I will create a new milestone called 1.0.x which will be used for issues that will need to be serviced to that 1.0 package

pgrawehr, 2020-10-16:
@joperezr : This issue and the milestone should be closed - they're both about the 1.0 release (erroneously called 3.0 here)

Ellerbach, 2020-10-16:
Thanks @pgrawehr, so I'll just close it. Feel free to reopen if it's really about 3.0 :-)

closed 2020-10-16 by Ellerbach