dotnet/iot #838: Add support for MotorHat

nahueltaibo, 2019-11-05:
I think the *MotorHat* for the Raspberry Pi is a pretty useful hat as it has the basic things you need to build robots. Adding support for this would allow simpler net core based robots.

Main page of this Hat, with all its details: [here](https://www.adafruit.com/product/2348)
It can also be bought in Aliexpress: [here](https://www.aliexpress.com/item/32857529077.html)

This Hat is based on the Pca9685, so dotnet iot repository already has the basic needed to support it.

I already have a basic version of the implementation need, based on the already available Pca9685 device binding.

If you think that makes sense, I'd like to contribute pushing this implementation at least as a starting point to be reviewed or improved.

krwq, 2019-11-05:
@nahueltaibo I think in general it might make sense. If hat is just Pca9685 connected to I2C or similar I'd just mention it and add a sample on the README.md pages.

If there is more than just Pca9685 then it might make sense to create a binding.

Quick look tells me it also adds infrared receiver and perhaps more.

Each device which is on the board if possible should be added separately and then one device collecting capabilities can be created as a binding.

For a sample of multi-device hat see [Sense Hat](https://github.com/dotnet/iot/blob/master/src/devices/SenseHat/SenseHat.cs)

nahueltaibo, 2019-11-06:
@krwq  Thanks for your feedback.

I don't think the MotorHat has a IR receiver. (at least not the one in the links I sent)
I checked the Sense Had code, and It does seem that the Motor Hat would follow the same idea.

I started the Motor Hat implementation here: [https://github.com/nahueltaibo/iot/tree/feature/motorhat/src/devices/MotorHat](https://github.com/nahueltaibo/iot/tree/feature/motorhat/src/devices/MotorHat)

It currently supports adding the 4 DCMotors that the device supports. For that, I extended Iot.Device.DCMotor with my own implemetation: *[DCMotor3Pwm](https://github.com/nahueltaibo/iot/blob/feature/motorhat/src/devices/MotorHat/DCMotor3Pwm.cs)*

I'm not sure another static CreateDCMotor3Pwm(...) should be added to Iot.Device.DCMotor as this implementation is a pretty uncommon one (using 3 pwm outputs for just one motor)

Do you think I'm going in the right direction?
Should I keep working with the idea of adding this as a device binding? (or should I create this as a separate nuget package on my own, that just uses the dotnet iot packages)

krwq, 2019-11-06:
@nahueltaibo looking at the code I think it makes sense to keep the DCMotor3Pwm separate considering usage of PWM in there is rather weird. If we have more similar devices we can move it there.

krwq, 2019-11-07:
@nahueltaibo thanks! (btw next time you can put `Fixes #838` as a first line of PR description so that the issue gets auto-closed

closed 2019-11-07 by krwq

nahueltaibo, 2019-11-07:
Thanks!
I'll try y that next time