dotnet/iot #844: DC Motor Sample Speed

acarlonvs, 2019-11-10:
_I am using a custom motor, so not sure if this is just the behaviour of my motor and there is something in the pwm libraries that expects this behaviour or if this is some kind of a convention. Raising here incase it is a bug._

_As a workaround, I am using SoftwarePwmChannel directly and that seems to have solved it for me._ 

**Describe the bug**

DC Motor sample appears to have incorrect reverse speed. Specifically, iot/src/devices/DCMotor/DCMotor2PinNoEnable.cs there is the line: 

    `_pwm.DutyCycle = 1.0 + val;`

when the provided speed is zero. Speed is between -1 and 1. If less than 0 it should set direction pin high, but I would assume that speed would just be the absolute value of whatever was passed in. E.g. if speed passed in was -0.2, I would expect direction pin to go high and the speed (duty cycle) to be 0.2. In the current code shown above, the duty cycle would be 0.8. In my test this was causing unexpected speed on reverse (given that I am using a custom motor). 

**Steps to reproduce**

Using DCMotor, set Speed to negative. 

```csharp
public void Go()
        {
            int pinFault = 5;
            int pinSlip = 22;
            Stopwatch sw = Stopwatch.StartNew();
            // 1 pin mode
            // using (DCMotor motor = DCMotor.Create(6))
            // using (DCMotor motor = DCMotor.Create(PwmChannel.Create(0, 0, frequency: 50)))
            // 2 pin mode
            // using (DCMotor motor = DCMotor.Create(27, 22))
            // using (DCMotor motor = DCMotor.Create(new SoftwarePwmChannel(27, frequency: 50), 22))
            // 3 pin mode
            // using (DCMotor motor = DCMotor.Create(PwmChannel.Create(0, 0, frequency: 50), 23, 24))
            using (GpioController controller = new GpioController())
            using (var motor = DCMotor.Create(12, 24, controller))
            {
                controller.OpenPin(pinFault, PinMode.Output);
                controller.OpenPin(pinSlip, PinMode.Output);
                controller.Write(pinFault, PinValue.High);
                controller.Write(pinSlip, PinValue.High);
                bool done = false;
                Console.CancelKeyPress += (o, e) =>
                {
                    done = true;
                    e.Cancel = true;
                };
                int dir = 1;
                while (!done)
                {
                    var speed = 0.2;                   
                    motor.Speed = dir * speed;
                    Console.WriteLine($"Speed:{motor.Speed}");
                    Thread.Sleep(100);
                    dir *= -1;
                }
            }            
        }
```

**Expected behavior**

-0.2 to be same duty cycle as forward 0.2, but different direction.

**Actual behavior**

0.2 speed = 0.2 duty cycle 
-0.2 speed = 0.8 duty cycle 

**Versions used**

1.0.19.47001

krwq, 2019-11-13:
@acarlonvs I think the logic is correct (AFAIR I have tested this). Note that once you reverse polarity then you expect 0.2 time to be spent "high" which is equivalent to 0.8.

Can you show me how is your DC Motor connected?

This specific setup expects that both pins directly control the value of the voltage (i.e. 1, 0 will turn it in one direction and 0, 1 in other)

krwq, 2023-01-19:
[Triage] Closing because this seems to be correct, please re-open if that is in fact an issue

closed 2023-01-19 by krwq