dotnet/iot #846: Iot.Device.Media does not exist

clevelandcs, 2019-11-12:
**Describe the bug**

I am unable to access Iot.Device.Media and follow the [VideoDevice](https://github.com/dotnet/iot/tree/master/src/devices/Media/VideoDevice) example. I'm working on a senior project using image processing to trigger GPIO signals on a Raspberry Pi 3/4 using still images captured from USB webcams GPIO library is working great but @ZhangGaoxing 's PR merged on Sept 30th which looks like it should have been included in 1.0.0 is not.

**Steps to reproduce**

```csharp
using Iot.Device.Media;
public class Camera
{
private void CaptureStill()
        {
            VideoConnectionSettings settings = new VideoConnectionSettings(busId: 0, captureSize: (2560, 1920), pixelFormat: PixelFormat.YUYV);
            using VideoDevice device = VideoDevice.Create(settings)
            {
                // Capture static image
                device.Capture("/home/pi/jpg_direct_output.jpg");

                // Change capture setting
                device.Settings.PixelFormat = PixelFormat.YUV420;

                // Get image stream, convert pixel format and save to file
                MemoryStream ms = device.Capture();
                Color[] colors = VideoDevice.Yv12ToRgb(ms, settings.CaptureSize);
                Bitmap bitmap = VideoDevice.RgbToBitmap(settings.CaptureSize, colors);
                bitmap.Save("/home/pi/yuyv_to_jpg.jpg", System.Drawing.Imaging.ImageFormat.Jpeg);
            }
        }
}
```

**Expected behavior**
1. Namespace Iot.Device.Media is recognized
2. VideoDevice and VideoConnectionSettings types exist
3. USB video device using v4l-utils can capture a still image
Describe what did you expect to happen

**Actual behavior**
The type or namespace Media does not exist in Iot.Device and expected types do not exist
**Versions used**
"Iot.Device.Bindings" Version="1.0.0"
"System.Device.Gpio" Version="1.0.0"
**Build:**
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18362
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.0.100\

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  2.1.801 [C:\Program Files\dotnet\sdk]
  2.2.203 [C:\Program Files\dotnet\sdk]
  3.0.100 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
**Run (Pi 3):**
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /opt/dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  2.1.802 [/opt/dotnet/sdk]
  3.0.100 [/opt/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.13 [/opt/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.13 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.13 [/opt/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [/opt/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download

krwq, 2019-11-13:
@clevelandcs I think we might have locked the branch before it was merged. Note I cannot see it anywhere in here:
https://github.com/dotnet/iot/tree/release/3.0/src/devices

In order to use it I'd recommend using these Nuget feeds:
https://github.com/dotnet/iot#net-cli

krwq, 2019-11-26:
@clevelandcs I'm closing the issue since we are unlikely to port this device to 1.0 at this point and you can use the prerelease nuget feeds to unblock yourself. Feel free to open/comment if you have any more questions here.

closed 2019-11-26 by krwq