dotnet/iot #849: Can't run Mcp25xxx sample on win-x64

scout208, 2019-11-13:
Trying to build and run the device sample for Mcp25xxx located in src\devices\Mcp25xxx\samples on a 64-bit Windows 10 computer gives the following output: 

````
C:\Users\anime\Documents\GitHub\iot\src\devices\Mcp25xxx\samples>dotnet run -r win-x64
Hello Mcp25xxx Sample!
Reset Instruction

Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'libc' or one of its dependencies: The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at Interop.open(String pathname, FileOpenFlags flags)
   at System.Device.Spi.UnixSpiDevice.Initialize() in C:\Users\anime\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line 58
   at System.Device.Spi.UnixSpiDevice.WriteByte(Byte value) in C:\Users\anime\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line 163
   at Iot.Device.Mcp25xxx.Mcp25xxx.Reset() in C:\Users\anime\Documents\GitHub\iot\src\devices\Mcp25xxx\Mcp25xxx.cs:line 203
   at Iot.Device.Mcp25xxx.Samples.Program.Reset(Mcp25xxx mcp25xxx) in C:\Users\anime\Documents\GitHub\iot\src\devices\Mcp25xxx\samples\Mcp25xxx.Sample.cs:line 53
   at Iot.Device.Mcp25xxx.Samples.Program.Main(String[] args) in C:\Users\anime\Documents\GitHub\iot\src\devices\Mcp25xxx\samples\Mcp25xxx.Sample.cs:line 26
````

This is the device I'm using to run and build the sample project: 

````
C:\Users\anime\Documents\GitHub\iot\src\devices\Mcp25xxx\samples>dotnet --info
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18362
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.0.100\

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  2.1.802 [C:\Program Files\dotnet\sdk]
  3.0.100 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
````

The MCP2515 device is connected to this computer over SPI.

pgrawehr, 2019-11-13:
This doesn't look right if running on Windows:
```
    at System.Device.Spi.UnixSpiDevice.Initialize() in C:\Users\anime\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line 58
   at System.Device.Spi.UnixSpiDevice.WriteByte(Byte value) in C:\Users\anime\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line 163
``` 
It looks like you built the wrong configuration. 
(Out of curiosity: What hardware do you use to add an SPI bus to a Windows PC?)

scout208, 2019-11-13:
Yeah, I didn't think that looked right either. I know I specified the runtime to be win-x64 so I don't understand why it would be using the unix version of SpiDevice. 

I'm using the Up board from Aaeon with Windows 10 installed (https://up-board.org/).

krwq, 2019-11-13:
cc: @joperezr any clues? When I `dotnet publish -r win-x64` or `win10-x64` I'm getting right assets on my box but I don't have any x64 box which has SPI

@scout208 can you try self-contained app and see if it still repros? (`dotnet publish -r ...` and run the exe)

scout208, 2019-11-13:
I decided to switch over and try it on a raspberry pi 3 model b with windows iot core to see if it would work there by publishing as win-arm and got the same result when I tried to run it there:

````
[169.254.226.9]: PS C:\Sample> ./MCP25xxx.Samples.exe
Hello Mcp25xxx Sample!
Reset Instruction
./MCP25xxx.Samples.exe :
    + CategoryInfo          : NotSpecified: (:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError

Unhandled Exception:

System.DllNotFoundException: Unable to load DLL 'libc' or one of its dependencies: The specified module could not be
found. (Exception from HRESULT: 0x8007007E)
   at Interop.open(String pathname, FileOpenFlags flags)
   at System.Device.Spi.UnixSpiDevice.Initialize() in
D:\Users\ConnorEarly\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line
58
   at System.Device.Spi.UnixSpiDevice.WriteByte(Byte value) in
D:\Users\ConnorEarly\Documents\GitHub\iot\src\System.Device.Gpio\System\Device\Spi\Devices\UnixSpiDevice.Linux.cs:line
163
   at Iot.Device.Mcp25xxx.Mcp25xxx.Reset() in
D:\Users\ConnorEarly\Documents\GitHub\iot\src\devices\Mcp25xxx\Mcp25xxx.cs:line 203
   at Iot.Device.Mcp25xxx.Samples.Program.Reset(Mcp25xxx mcp25xxx) in
D:\Users\ConnorEarly\Documents\GitHub\iot\src\devices\Mcp25xxx\samples\Mcp25xxx.Sample.cs:line 53
   at Iot.Device.Mcp25xxx.Samples.Program.Main(String[] args) in
D:\Users\ConnorEarly\Documents\GitHub\iot\src\devices\Mcp25xxx\samples\Mcp25xxx.Sample.cs:line 26
````

That was running as a self-contained app there. Do you have a Raspberry Pi you can test on?

krwq, 2019-11-13:
Per my offline conversation @joperezr - when running the sample on Windows try copy pasting the sample outside of the repo and make it depend on the nuget packages instead of project to project references (we have some weirdness in the repo where only Linux works)

krwq, 2019-11-13:
The weirdness: https://github.com/dotnet/iot/blob/master/src/devices/Mcp25xxx/samples/Mcp25xxx.Samples.csproj#L12

another thing is I'm not sure what's the status of Mcp25xxx binding (cc: @shaggygi). I'd recommend using Linux with [SocketCan](https://github.com/dotnet/iot/tree/master/src/devices/SocketCan) since this was more tested

joperezr, 2019-11-13:
Specifically the problem is here:

https://github.com/dotnet/iot/blob/bd993d8d2e0919a6e9e41ce6e3b968fede35aa42/src/devices/Mcp25xxx/samples/Mcp25xxx.Samples.csproj#L11-L13

If you see we are passing in manually the Runtime Identifier. If you locally make the following changes, everything should work:

```xml
<ItemGroup>
  <PackageReference Include="System.Device.Gpio" Version="1.0.0" />
</ItemGroup>
```

That will make the right assets to get copied over to the output folder.

krwq, 2019-11-26:
@scout208 I think the Mcp25xxx is not fully finished yet at least looking at the sample: https://github.com/dotnet/iot/blob/master/src/devices/Mcp25xxx/samples/Mcp25xxx.Sample.cs
I also recall this device was rather complicated so we decided to merge working part (reading registers etc and incrementally fix it in the future)

As I mentioned above use SocketCan instead for now (Linux required) which should cover Mc25xxx.

Ellerbach, 2021-07-01:
[Triage] @shaggygi  as you are the original author of the MCP25xx, would be great if you can have a look at it.

krwq, 2023-01-19:
[Triage] This is most likely outdated, please re-open if the issue persists

closed 2023-01-19 by krwq