dotnet/iot #850: DhtSensor DhtType.Dht11 always returns NaN on .net core running a GrovePi+ on a Rasbperry Pi with Rasbpian OS

JodyPetroni, 2019-11-14:
I am trying to get the temperature and humidity sensor working on .net core running a GrovePi+ on a Rasbperry Pi with Rasbpian OS. This code always produces NaN output.

```
Console.WriteLine("Hello GrovePi!");
            
I2cConnectionSettings i2CConnectionSettings = new I2cConnectionSettings(1, GrovePi.DefaultI2cAddress);
grovePi = new GrovePi(I2cDevice.Create(i2CConnectionSettings));

Console.WriteLine($"Manufacturer :{grovePi.GrovePiInfo.Manufacturer}");
Console.WriteLine($"Board: {grovePi.GrovePiInfo.Board}");
Console.WriteLine($"Firmware version: {grovePi.GrovePiInfo.SoftwareVersion}");
DhtSensor dhtSensor = new DhtSensor(grovePi, GrovePort.DigitalPin4, DhtType.Dht11);
while (true)
 {
       dhtSensor.Read();
       Console.WriteLine($"{dhtSensor.DhtType}: {dhtSensor}");

       Thread.Sleep(2000);
 }
```

kurianov-pavel, 2019-11-19:
@JodyPetroni Try to use something like that
`
var dht11 = new Dht11(YOUR_PIN, PinNumberingScheme.Logical);

                if (!dht11.IsLastReadSuccessful)
                    return result;

                result.Data = new DthData
                {
                    Humidity = dht11.Humidity,
                    Temperature = dht11.Temperature.Celsius
                };
                result.IsSuccess = true;
`

and you can add delay time. Sometimes it helps (like one read data every/per 2 seconds)

JodyPetroni, 2019-11-19:
alas same result

kurianov-pavel, 2019-11-20:
> alas same result

Do you use it?

` using (GpioController controller = new GpioController(PinNumberingScheme.Logical))
            {YOUR_GPIO_CODE}`

JodyPetroni, 2019-11-20:
I don’t follow. What do I do with controller?

kurianov-pavel, 2019-11-20:
> I don’t follow. What do I do with controller?

Something like this
```
public static async Task Main(string[] args)
        {
            using (GpioController controller = new GpioController(PinNumberingScheme.Logical))
            {
                var dht11 = new Dht11(YOUR_PIN, PinNumberingScheme.Logical);
                isCancel = false;

                while (!isCancel)
                {
                    Console.WriteLine($"--- Humidity = {dht11.Humidity} ---");
                    Console.WriteLine($"--- Humidity = {dht11.Temperature.Celsius} ---");
                }
                
                Console.ReadLine();
            }
        }
```

joperezr, 2019-11-20:
you don't need a GpioController. The code that @JodyPetroni has looks correct to me and matches the sample we have for the GrovePi. The only thing that doesn't match the sample is the GrovePi Digital Pin, are you sure you have everything wired up correctly? Here is what the sample code has:

https://github.com/dotnet/iot/blob/f92c6fcf4dcedaa5911c5ab8b0d3ff6ad0856881/src/devices/GrovePi/samples/Program.cs#L34

And here is the photo of how the GrovePi is connected : https://github.com/dotnet/iot/blob/master/src/devices/GrovePi/samples/sample.jpg 

Make sure you are talking to the right port and digital pin.

cc: @Ellerbach who originally wrote the GrovePi binding.

JodyPetroni, 2019-11-21:
@joperezr @Ellerbach I tried all Digital Pins with same result. I have also tried the python code and it works as expected.

krwq, 2019-11-22:
@JodyPetroni can you try running python GrovePi example as a sanity check? Also try running the same app just after you tried Python (I recall seeing issue in the past that something had to be installed from there to get this to work in .NET) - if that will fix it we should take a deeper look on what exactly is being installed and if we can permanently fix that

JodyPetroni, 2019-11-22:
I ran the python code and it worked
then ran .net code and it didnt
`dotnet build --runtime linux-arm
dotnet run --runtime linux-arm`

krwq, 2019-11-22:
@JodyPetroni can you try running as sudo? If still doesn't work then this should be taken closer look at

JodyPetroni, 2019-11-24:
[image]

Ellerbach, 2019-11-26:
I will try as soon as I can to check.

Ellerbach, 2019-11-26:
So just checked the code and it's working all good for me:
[image]

I've been using the example code that Jose pointed out: https://github.com/dotnet/iot/issues/850#issuecomment-556095669
I removed all the sensors I'm not using. See below the code.

Here is the hardware setup:
[image]

Here is the full code:
```csharp
using Iot.Device.GrovePiDevice;
using Iot.Device.GrovePiDevice.Models;
using Iot.Device.GrovePiDevice.Sensors;
using System;
using System.Device.Gpio;
using System.Device.I2c;
using System.Threading;

namespace dhtgrove
{
    class Program
    {

        static private GrovePi grovePi;

        static void Main(string[] args)
        {
            Console.WriteLine("Hello GrovePi!");
            PinValue relay = PinValue.Low;
            I2cConnectionSettings i2CConnectionSettings = new I2cConnectionSettings(1, GrovePi.DefaultI2cAddress);
            grovePi = new GrovePi(I2cDevice.Create(i2CConnectionSettings));
            Console.WriteLine($"Manufacturer :{grovePi.GrovePiInfo.Manufacturer}");
            Console.WriteLine($"Board: {grovePi.GrovePiInfo.Board}");
            Console.WriteLine($"Firmware version: {grovePi.GrovePiInfo.SoftwareVersion}");
            Console.ReadKey();
            DhtSensor dhtSensor = new DhtSensor(grovePi, GrovePort.DigitalPin7, DhtType.Dht11);
            while (!Console.KeyAvailable)
            {
                Console.Clear();
                dhtSensor.Read();
                Console.WriteLine($"{dhtSensor.DhtType}: {dhtSensor}");
                Thread.Sleep(2000);
            }
        }
    }
}
```

From the PC, I do compile using:
dotnet publish -r linux

Then I copy the files on the Rapsberry using WinScp:
[image]

Then I connect in ssh on the Raspbery Pi. And I do change the ./dthgroove permission to execute it using ```chmod +x ./dthgroove```

Then I just run ```./dthgroove```, here are the firmware and type of my GRovePi:
[image]

And I get after 1 measure a correct temperature and humidity.
[image]

Ellerbach, 2019-11-26:
@JodyPetroni I will update the firmware to the 1.4.0 version as well and test with this version. For compatibility reason, I've been using the 1.2.2 to make sure all is working even with older versions. 
That said the DHT code seems not to have changed over time. I'll let you know the result as well

Ellerbach, 2019-11-26:
@JodyPetroni after upgrading to 1.4.0, it seems than DHT sensor is not working and has the same behavior as you describe. I will investigate to correct the issue. If you have an urgent need, please downgrade to version 1.2.2 which is for sure working. I've tested as well with 1.2.7 and it's working.
Thanks for your tests and your patience.

Ellerbach, 2019-11-26:
@JodyPetroni, so I found the issue and it's related to the way the DHT sensor is managed now on this new firmware.

You can use this code the time I make a proper PR to fix this in the main repos and the time a new binding is available:
```csharp
            grovePi.WriteCommand(GrovePiCommand.DhtTemp, GrovePort.DigitalPin4, (byte)DhtType.Dht11, 0);
            while (!Console.KeyAvailable)
            {                
                var retArray = grovePi.ReadCommand(GrovePiCommand.DhtTemp, GrovePort.DigitalPin7);
                if (retArray[0] != 0x17)
                {
                    grovePi.WriteCommand(GrovePiCommand.DhtTemp, GrovePort.DigitalPin4, (byte)DhtType.Dht11, 0);
                    var temp = BitConverter.ToSingle(retArray.AsSpan(1, 4));
                    var hum = BitConverter.ToSingle(retArray.AsSpan(5, 4));
                    Console.WriteLine($"Temp: {temp.ToString("0.00")} °C, Hum: {hum.ToString("0.00")} %");
                }
                else
                {
                    Console.WriteLine("No temperature or humidity available");
                }
                Thread.Sleep(2000);
            }
```

Here is the result:
[image]

JodyPetroni, 2019-11-26:
Thanks will update with your code - should the write command be to pin7 and not pin4?

JodyPetroni, 2019-11-26:
Hi I can confirm above code change works after changing pin4 to pin7 (where the sensor is attached to pin 7 in my case)

Ellerbach, 2019-11-27:
@JodyPetroni thanks for the confirmation. I expect to work on a PR fix next week. You'll be able to use better code than the one posted here. But in the mean time, depending on what you need to do, that should work fine.

joperezr, 2019-12-09:
@Ellerbach now that your PR is merged I assume that we can go ahead and close this issue?

Ellerbach, 2019-12-10:
> @Ellerbach now that your PR is merged I assume that we can go ahead and close this issue?

Sure, it is closed as it's now corrected and merged. Thanks

krwq, 2019-12-12:
FYI: if you put `Fixes #850` on the first line of the PR description it will auto-close the issue when the PR is merged which makes tracking much easier

closed 2019-12-12 by krwq

avongonzales, 2020-04-28:
Hello
I have RaspberryPI 3 and using Latest Raspbian "Buster"
I updated the GrovePi+ Firmware to  version: 1.4.0

Do you have Code Samples for .NET Core3.1 ?
Grove-LCD_RGB_Backlight   http://wiki.seeedstudio.com/Grove-LCD_RGB_Backlight/  
Grove - Red LED http://wiki.seeedstudio.com/Grove-Red_LED/  
Grove Buzzer - http://wiki.seeedstudio.com/Grove-Buzzer/

Thanx

Ellerbach, 2020-04-29:
@avongonzales , see also https://github.com/dotnet/iot/tree/master/src/devices/CharacterLcd for the Grove LCD. It's a classical I2C display supported in a different binding.

avongonzales, 2020-04-29:
Thank you K.W &  L.E.. Cheers !!!