dotnet/iot #857: Support setting the GPIO Pin mode to ALT(n) on the Raspi

pgrawehr, 2019-11-21:
The mode of each GPIO Pin can be set to IN, OUT or up to 5 alternate modes (ALT0-ALT5). For instance, BCM Pin 12 can be set to IN, OUT, PWM0 (ALT0), SMI (ALT1), DPI (ALT2), AVEOUT (ALT3), AVEIN (ALT4) or JTAG TMS (ALT5). See the BCM Chip reference documentation or https://pinout.xyz/ for the alternate functions of each pin. Note that this page was not yet updated for the PI4, which has even more alternate functions (one of them being 4 different I2C channels spread over different pins, see prelimary datasheet, page 10: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf). 

By default, all pins are set as input after a reboot, except those for which a driver module is loaded (i.e. if the i2c overlay is loaded, Pins 2 and 3 are set to ALT0 after startup). The current mode of each pin can for instance be queried using the `gpio` tool by entering `gpio readall` at the command line. Nothing prevents changing the pin mode to anything else after bootup though. So even if the i2c module is loaded, setting pin 2 to output is still possible, but of course i2c commands won't work after that.

The library is currently only supporting setting the modes to input or output (together with the pullups, but that setting is independent, even though it only makes sense for an input pin). If once set to output, there's no way of setting it back to ALT0 mode which is required for I2C, except a reboot or a manual command using an alternate library (i.e. `gpio -g mode 2 ALT0` will do).

While it usually will not make much sense to change the mode of the I2C pins (if an I2C device is in use) a programming error may still lead to such a wrong setting during development or after a hardware reconfiguration by the user. Some modes may even be used interchangeably by the same program, for instance OUT and PWM for driving a LED. 

**Suggestion**
Extend PinMode to add members for ALT0-ALT5. Implement the setting of these modes in RaspberryPiDriver (should be straight-forward, is only a different combination of mode bits in the same registers as the Input/Output mode is set). 
For drivers or hardware where these modes are not supported, NotSupportedException is thrown (same as happens now if setting mode to InputPullUp if that setting is not supported).

krwq, 2019-11-21:
@pgrawehr I actually think of GPIO as pure input/output with or without pull up/down.

The alternative functions should not be exposed directly IMO. Rather than that I think GpioController should have something like:

`PwmChannel CreatePwmChannel(int pin)`/`IEnumerable<int> GetPwmPinsWithHardwareSupport` and similar for other functions which will set the mode to the correct alternative form under hood.

Implementation can use Alt0-5 wherever they want but for users I think the API should be much simpler and they should not need to know what Alt means.

pgrawehr, 2019-11-22:
@krwq That's a feasible option, however it would still need an update of the PinMode enum I think, because `GetPinMode` needs to return something meaningful if the pin is in one of the Alt modes (even if it's just `PinMode.Other` or something).

krwq, 2019-11-22:
@pgrawehr `Other` or similar sounds reasonable to me. But let's wait until we have more opinions/upvotes here before we implement it.

krwq, 2019-12-04:
As per discussion in the other thread this issue should have some follow up work since in some cases as in #874 lack of this is causing PWM to become unusable: setup one of the pins as PWM through /boot/config.txt and then using SysFs or RaspberryPiDriver change pin to input/output. Driver is not aware of the mode change and that effectively causes PWM to not work.

We should have some options of preventing this since this is not particularly obvious issue. Some options:
- throw an exception by default when user tries to change mode from something we do not understand - possibly give user a way of expressomg "change mode back to whatever it was"
- PWM could be aware of GPIO and ensuring that mode is set correctly - this has somewhat a layering issue - currently PWM code is independent of GPIO code - assuming we do that which driver should PWM prefer?

pgrawehr, 2019-12-04:
I think the "Board" approach would solve most of the issues we have.

Let's create a class hierarchy with a base class like:
```
public abstract class Board
{
  public I2cDevice CreateI2cDevice(int channel, int[] pins); 
  public PwmChannel CreatePwmChannel(Settings..., int pin);
  public GpioController CreateGpioController(...);
  public Temperature GetCPUTemperature();
  // ...
}
``` 
Then the specific Create Method can "allocate" the pin for the given service and also make sure that it is set to the proper mode. For full verification, the desired pin needs to be provided, because sometimes, the same device can be linked to several different pins (i.e. on the Raspi, PWM Channel 0 can be routed to pins 12 or 18) and we need to know which one to configure (for PWM in this case). If the GpioController is also created trough this class (or it's board-specific implementation), it can get a backlink to the board to make sure pins are only opened in one mode at a time. 

If we're going to implement firmata support (#262), something like this is required as well, as we - probably - want to provide the connection information only once, and not each time we create a pwm device or an I2C device. Plus he doesn't want to explicitly create an `ArduinoPwmChannel` but just a `PwmChannel` on the current board. Like this, only the initialisation code needs to be changed slightly to run the same code on different boards (assuming they're connected to the same hardware on the GPIO pins, of course).

krwq, 2019-12-04:
@pgrawehr I'm not sure abstracting this will always work. Note that some boards will have thermometers (most will likely have some sort of CPU thermometer), some might have different kinds of sensors, some might have ADC/DAC, some won't. I kinda think of board as being rather unique but each of them providing certain set of capabilities or devices/subdevices. Each board can be designed completely uniquely and therefore I'm unsure if class hierarchy will work.
Also @Ellerbach is currently prototyping a solution which allows using many devices with regular USB port which will work also on x64/x86 and board will not exactly work here as well.

I think we should create a separate issue for this design conversation and here focus on requirements rather than specific solution. Once we have couple of issues like this one we can then justify the design reasonably based on the requirements.

P.S. (about Firmata) We currently have a [similar capability pending in PR](https://github.com/dotnet/iot/pull/875)

Ellerbach, 2021-07-01:
[Triage] This is now implemented.

closed 2021-07-01 by Ellerbach