dotnet/iot #860: hcsr04 not working

mringl, 2019-11-21:
I've a problem running the HCSR04 device on a Pi3 with dotnet core 3.1 preview 2.
I'll never get out of the first while(_controller.Read(_echo) == PinValue.Low)
If I read from _echo after _controller.Write(_trigger, PinValue.High); everything works fine.

I've also confirmed my setup via a python script which works correctly.

krwq, 2019-11-22:
Thanks for the report. Can you try running the same code as sudo and see if still repros? Also do you have libgpiod installed?

I think except for fixing the issue we should also add a protection against endless looping here.

@joperezr this is another place where it would be useful to know which GpioDriver does the GpioController use since this is really hard to diagnose otherwise

mringl, 2019-11-23:
Hi, sudo doesn't help, I've also now libgpiod installed but no luck.
I've also tried to use the _controller.WaitForEvent method instead of the loop, no luck.
By the way it is a Rasp 2 B not 3

krwq, 2019-11-26:
interesting, it could be related to Raspberry Pi 2, we might be using different driver (sysfs).

The eventing is kinda broken on sysfs driver (we currently are not sure if it's possible to fix it). We might be able to fixup the RaspberryPi3 driver though to work with RPi2. I'm surprised installing libgpiod hasn't fixed it, this is something we should look into.

One more thing to try: can you create GpioController using following line:
`new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver())` and pass it to the HCSR04

and see if you have any luck or perhaps some more meaningful error message?

Kash0321, 2019-12-29:
Same issue here. Is a Raspberry Pi 3B+ in my case... Tested with Python sensor works properly, but not with .NET Core / .NET IoT

I'd tried your last advice (new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver())) with no luck...

Kash0321, 2019-12-29:
I've found the problem right now, and it isn't a drivers issue... Problem is on Hcsr04 class: https://github.com/dotnet/iot/blob/master/src/devices/Hcsr04/Hcsr04.cs

In line 63, TimeSpan.FromMilliseconds(0.01) returns 0, then, no pulse is sended nor received by the sensor...

            // Trigger input for 10uS to start ranging
            _controller.Write(_trigger, PinValue.High);
            Thread.Sleep(TimeSpan.FromTicks(10)); <---- With this change, now it works
            _controller.Write(_trigger, PinValue.Low);

Kash0321, 2019-12-30:
My collegue @rsciriano has done some investigation about the issue and has found this: https://github.com/dotnet/coreclr/issues/26063

It seems that "someone" has broken the behavior of the FromMilliseconds method...

It also seems that, for now, the best option is to use FromTicks instead of FromMilliseconds... I will try to create a PR to fix the issue for de Hcsr04 binding

krwq, 2019-12-31:
Since this seems to be related to double rounding perhaps we should temporarily use TimeSpan.FromTicks instead to avoid any rounding issues... other option is to use DelayHelper

krwq, 2019-12-31:
Actually quickly debugging seems to me that first call of `_controller.Read` is taking much longer than expected and we miss initial echo and get stuck, If I call it at least once before I use the sensor it works correctly for me. I'll send a patch and let you test against it

krwq, 2019-12-31:
Ok, I've sent a patch, I'd appreciate if you've tried it on your setups and see if this fixes all the issues (including double checking if read values are correct)

Kash0321, 2019-12-31:
OK, let me try it out in the next couple of days... Thank you very much, and happy new year!

krwq, 2019-12-31:
Happy new year!

Kash0321, 2020-01-01:
I've tested your code in my setup and it works perfectly. However, I have a couple of considerations:

- The problem with the "FromMilliseconds" method still exists, but it only repros when the code is executed on .Net Core 2.2....    **On .Net Core 3.1 works fine**.
- However, now it seems to be irrelevant, because of the protections you have implemented

Thanks you very much

closed 2020-01-02 by krwq