dotnet/iot #863: Platform not supported exception if using PowerShell Import-Module

DanielSSilva, 2019-11-22:
I have a C# DLL that has the `System.Device.Gpio Version="1.0.0"` and I want to use it with PowerShell, through Import-Module. Upon importing the DLL, if I try to create an I2cConnectSettings object, I get an exception saying that `Operation is not supported on this platform`.
On the same project, I've created the Main method that does exactly the same. If I run `dotnet donetIot_test.dll` the code runs successfully.

I can't seem to find the root cause of the exception, but the platform is the same (raspberry pi 3b) - assuming that the platform is the hardware.

If I run the `dotnet donetIot_test.dll` command _inside_ PowerShell it works, so it doesn't seem to be shell-related. 

[Here's a link to the repo with a sample](https://github.com/DanielSSilva/dotnet_iot_powershell_sample)

**Versions used**

Add following information:

<details>
<summary>dotnet --info on the machine being used to build</summary>
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     Mac OS X
 OS Version:  10.13
 OS Platform: Darwin
 RID:         osx.10.13-x64
 Base Path:   /usr/local/share/dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  2.1.202 [/usr/local/share/dotnet/sdk]
  2.1.700 [/usr/local/share/dotnet/sdk]
  2.2.107 [/usr/local/share/dotnet/sdk]
  3.0.100 [/usr/local/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.2.5 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.2.5 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.0.9 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.5 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
</details>

<details>
<summary> dotnet --info on the machine where app is being run (not applicable for self-contained apps) </summary>
pi@raspberrypi:~/PowerShell_IoT/publish $ dotnet --info
  It was not possible to find any installed .NET Core SDKs
  Did you mean to run .NET Core SDK commands? Install a .NET Core SDK from:
      https://aka.ms/dotnet-download

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  No SDKs were found.

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0 [/home/pi/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
</details>

- Version of `System.Device.Gpio` package: 1.0.0

DanielSSilva, 2019-11-22:
Not sure if this helps to understand
[image]

joperezr, 2019-11-22:
We would need more info in order to investigate this (like callstack, or the way you published the app), have you tried breaking this up in a debugger? The reason why I believe this is happening is because we have 3 different implementations for the library, one is for windows, one is for linux, and the other one is platform-agnostic which just throws platform not supported exception. I believe the problem here is that when you call the program via dotnet, the runtime loader knows how to read the json config files with the app, and will know how to pick the right version of System.Device.Gpio (in this case it should pick the windows implementation) but when you load it via powershell, powershell doesn't know how to read that runtime json file, so it just picks the platform agnostic which throws. In order to validate this, can you please try again, but this time publishing your app like: `dotnet publish -r win-x64`? This will make sure that when creating the publish directory, we will only select the assets that are relevant to windows x64, and won't publish anything else which will probably help powershell pick the right module to load

joperezr, 2019-11-22:
sorry just noticed that you are actually running on a raspberry pi, not in Windows, so what you want for a publish command is this: `dotnet publish -r linux-arm` This will make sure that we only publish the linux implementation so hopefully that should just work

DanielSSilva, 2019-11-22:
Yes! That was exactly the problem!
Since I saw the `runtimes` inside the publish folder, I didn't even try to compile with the Linux as runtime.
[image]

joperezr, 2019-11-22:
Closing as the issue seems to be resolved.

closed 2019-11-22 by joperezr